aerostack2/aerostack2 #103: [as2_python_api] Make behavior status shared

javilinos, 2022-12-21:
### Feature Description

When launching drone interface from python mission and then launching drone interface with same namespace from another source (eg: keyboard_teleoperation), the second one doesn't get the behaviors status from the first one. There is a workaround which is executing keyboard_teleoperation from the mission file with thread parameter=True. 

### Implementation Considerations

Make behavior status shared between both interfaces if they have same namespace.

pariaspe, 2022-12-21:
Please, add some explanations to reproduce this

javilinos, 2022-12-21:
Define mission with drone interface with namespace=drone_sim_0. Execute as2_keyboard_teleoperation with drone interface with namespace=drone_sim_0. Execute mission. Behavior status values of mission's drone interface are not returned in behavior status variables of as2_keyboard_teleoperation's drone interface.

pariaspe, 2022-12-22:
I'm testing the following script:
```python
import time
from enum import Enum

import rclpy

from as2_python_api.drone_interface import DroneInterface

class BehaviorStatus(Enum):
    IDLE = 0
    RUNNING = 1
    PAUSED = 2

rclpy.init()

drone_interface = DroneInterface("", verbose=False)
drone_interface_2 = DroneInterface("", verbose=False)

time.sleep(1)

print("STATUS:", BehaviorStatus(drone_interface.takeoff.status).name)
print("STATUS:", BehaviorStatus(drone_interface_2.takeoff.status).name)

print("START")
drone_interface.takeoff.start(1.0, 0.5, False)

time.sleep(2)
print("STATUS:", BehaviorStatus(drone_interface.takeoff.status).name)
print("STATUS:", BehaviorStatus(drone_interface_2.takeoff.status).name)

print("PAUSE")
drone_interface_2.takeoff.pause()

time.sleep(2)
print("STATUS:", BehaviorStatus(drone_interface.takeoff.status).name)
print("STATUS:", BehaviorStatus(drone_interface_2.takeoff.status).name)

print("RESUME")
drone_interface.takeoff.resume(False)
time.sleep(2)
print("STATUS:", BehaviorStatus(drone_interface.takeoff.status).name)
print("STATUS:", BehaviorStatus(drone_interface_2.takeoff.status).name)

print("WAIT TO RESULT")
drone_interface_2.takeoff.wait_to_result()

print(drone_interface.takeoff.result)
print(drone_interface_2.takeoff.result)

print("Exit")
drone_interface.shutdown()
drone_interface_2.shutdown()
```
where I instantiate two drone interfaces with the same namespace. Notice that namespace is empty because I'm testing it against as2_behavior `basic_test` node that doesn't have namespace.
First, I start takeoff behavior from instance_1 and pause it from instance_2. Then, I resume takeoff behavior from instance_1 again and wait for the result from instance_2. Meanwhile, I'm printing some stuff:

```
$ python multiple_behaviors.py 
[INFO] [1671706419.116803341] [_interface]: Starting 
[INFO] [1671706419.639850037] [_interface]:  interface initialized
[WARN] [1671706420.661901716] [_interface]: GoToWaypointBehaviour Not Available
[WARN] [1671706421.670935741] [_interface]: FollowPathBehaviour Not Available
[WARN] [1671706422.686862800] [_interface]: LandBehaviour Not Available
[WARN] [1671706422.687494856] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [1671706422.699512564] [_interface]: Starting 
[INFO] [1671706423.206568947] [_interface]:  interface initialized
[WARN] [1671706424.217313608] [_interface]: GoToWaypointBehaviour Not Available
[WARN] [1671706425.225900766] [_interface]: FollowPathBehaviour Not Available
[WARN] [1671706426.234448318] [_interface]: LandBehaviour Not Available
STATUS: IDLE
STATUS: IDLE
START
STATUS: RUNNING
STATUS: RUNNING
PAUSE
STATUS: PAUSED
STATUS: PAUSED
RESUME
STATUS: RUNNING
STATUS: RUNNING
WAIT TO RESULT
as2_msgs.action.TakeOff_Result(takeoff_success=True)
as2_msgs.action.TakeOff_Result(takeoff_success=True)
Exit
```

javilinos, 2022-12-22:
Tested with KB, works smooth.

closed 2022-12-22 by javilinos