aerostack2/aerostack2 #104: [as2_python_api] resume() blocks until behavior is finished

javilinos, 2022-12-21:
### ROS2 Version

humble

### Version

main

### What happened?

Uppon calling resume function of stopped behavior, behavior resumes its task successfully but it blocks until behavior finishes its task, so it cannot be stopped again. No error log is returned.

pariaspe, 2022-12-22:
Hi @javilinos , resuming can be done blocking or not:
```python
    def resume(self, wait_result: bool = True) -> bool:
        """Continue with current behavior

        :param wait_result: wait to behavior end, defaults to True
        :return: resume succeed or not
        """
```

javilinos, 2022-12-22:
Tested and works fine, thanks.

closed 2022-12-22 by javilinos