aerostack2/aerostack2 #107: [takeoff_behavior] <title>

pariaspe, 2022-12-22:
### ROS2 Version

humble

### Version

main

### What happened?

Launch crashes when adding option `-s (--show-arguments)`. 
Default values for `DeclareLaunchArgument` should be strings.

### Relevant log output

```shell
$ ros2 launch takeoff_behaviour takeoff_behaviour_launch.py -s
Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 54, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "/home/parias/as2_ws/install/takeoff_behaviour/share/takeoff_behaviour/launch/takeoff_behaviour_launch.py", line 44, in generate_launch_description
    DeclareLaunchArgument('takeoff_height', default_value=1.0),
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/actions/declare_launch_argument.py", line 122, in __init__
    self.__default_value = normalize_to_list_of_substitutions(default_value)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/normalize_to_list_of_substitutions_impl.py", line 44, in normalize_to_list_of_substitutions
    return [normalize(y) for y in cast(Iterable, subs)]
TypeError: 'float' object is not iterable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.3', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2launch/command/launch.py", line 170, in main
    return print_arguments_of_launch_file(launch_file_path=path)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2launch/api/api.py", line 123, in print_arguments_of_launch_file
    launch_description = get_launch_description_from_any_launch_file(launch_file_path)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 57, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: 'float' object is not iterable
```

pariaspe, 2023-01-09:
Other behaviors like Land, Goto or FollowPath do not have default values. Should we add them or remove it from Takeoff? @RPS98 @miferco97

RPS98, 2023-01-17:
Solved in #114

closed 2023-01-17 by RPS98