aerostack2/aerostack2 #132: [as2_aerial_platform] Management of time between set control mode and references updated

RPS98, 2023-01-18:
### ROS2 Version

humble

### Version

main

### What happened?

Current operation:
1º. Platform running with control mode A and references A'.
2º. Set Control Mode request arrives, and platform change control mode to B.
3º. Platform compute output with new control mode B, but with references A'.
4º. New reference B' arrives relative to new control mode B.
5º. Platform running with control mode B and references B'.

The 3º step needs to be avoid. Now, platforms get method ownSendCommand() called but there is no way to know if the new reference has been arrived or not.

RPS98, 2023-01-18:
Maybe, platforms should have a updateReference(...) method to manage this situation.
@miferco97 @pariaspe what do you think?

RPS98, 2023-12-19:
Aerial platform have the variable `has_new_references_` which indicates if this happens. It should be use on as2_aerial_platform or each platform should manage it?

miferco97, 2023-12-19:
what happens when no new_reference is received ?

RPS98, 2023-12-19:
The aerial platform calls the `own_send_command` method of the platform, but `this->command_pose_msg_`, `this->command_twist_msg_`, etc., are storing old control mode reference data.

We could wait until a new reference arrives and set `has_new_references_` to true. But during this elapse of time, `own_send_command` is not be calling.

miferco97, 2023-12-19:
maybe this behavior can be especified in the platform ?

RPS98, 2024-08-09:
Done in #584

closed 2024-08-09 by RPS98