aerostack2/aerostack2 #135: as2_msgs Drive for compliance with ROS REP 147

Ryanf55, 2023-01-19:
### Feature Description

Hello,

Just discovered the repo, seems like a great addition to the open source community. 

Have you considered trying to follow [ROS REP 147](https://ros.org/reps/rep-0147.html)? This would promote re-usability of the components here and allow interfacing with other systems without having to write converter nodes. 

### Implementation Considerations

I was curious what the design motivation for the messages in this package is. It seems like the aerial community in ROS2 hasn't yet decided upon a common set of messages for aerial applications. One approach is to mirror what is used in PX4, but there does not appear to be ABI stable in ROS releases. If, for example, I decided to use that package in a Humble project, [they are updated following the PC4 platforms update schedule](https://github.com/PX4/px4_msgs#how-are-these-messsage-definitions-generated), despite Humble being LTS. 

I am particularly interested in waypoint following. The NAV2 stack in ROS2 does not seem to support 3D navigation at this point, however the proposed interfaces in ROS REP 147 do. 

For example, the waypoint planning here is in a local ENU coordinate system, while the ROS rep uses LLA, while also includes velocity. 

Since this repo is relatively new, I thought it would be a good time to ask. If you are interested, I am happy to help with a proposal. Since ROS REP 147 is still in the deferred state, I think there is time to modify things, such as supporting actions, which is not currently done. 

Cheers!

pariaspe, 2023-01-23:
Hi @Ryanf55 , thanks for getting around.

Aerostack2 comes from a previous version [[1]](https://github.com/cvar-upm/aerostack/wiki) that had a slightly different philosophy. When the very first version of Aerostack was released, ROS REP147 wasn't written. We took the opportunity while moving to ROS2 to get closer to the ROS community.
So, answering your questions:
1. We are studying REP147 and we don't rule out using the standard in the future. Most of the interfaces described in REP147 are identical in the messages used (not in the topic names). Right now, we are focused in other aspects of the software.
2. The design motivation behind the messages is mostly to keep the messages used in our ROS1 version [[2]](https://github.com/cvar-upm/aerostack/wiki/Common-ROS-topics) that mainly rely on the ROS common messages. 
3. We used to fly indoors only and we are currently working on extending the software to outdoor (LLA) flights. See #134

We have a bunch of new ideas for the next months and we are happy to receive suggestions like yours.

Ryanf55, 2023-02-02:
> Hi @Ryanf55 , thanks for getting around.
> 
> Aerostack2 comes from a previous version [[1]](https://github.com/cvar-upm/aerostack/wiki) that had a slightly different philosophy. When the very first version of Aerostack was released, ROS REP147 wasn't written. We took the opportunity while moving to ROS2 to get closer to the ROS community. So, answering your questions:
> 
> 1. We are studying REP147 and we don't rule out using the standard in the future. Most of the interfaces described in REP147 are identical in the messages used (not in the topic names). Right now, we are focused in other aspects of the software.
> 2. The design motivation behind the messages is mostly to keep the messages used in our ROS1 version [[2]](https://github.com/cvar-upm/aerostack/wiki/Common-ROS-topics) that mainly rely on the ROS common messages.
> 3. We used to fly indoors only and we are currently working on extending the software to outdoor (LLA) flights. See [GPS behavior support #134](https://github.com/aerostack2/aerostack2/pull/134)
> 
> We have a bunch of new ideas for the next months and we are happy to receive suggestions like yours.

Thanks! I'm working on the ROS2 nav2 stack for waypoint navigation. Since REP-147 has a proposal for the messages related to this, please keep in the loop. Adopting a more common interface for waypoint navigation would be very handy to try out different planners. 

Keep in touch when you are ready to consider it.