aerostack2/aerostack2 #140: [as2_ign_gazebo_assets] Launch external objects with bridges

javilinos, 2023-01-25:
### Feature Description

Update as2_ign_gazebo_assets so it can launch external models that also need to publish data to ROS2. (eg: windmill).

### Implementation Considerations

as2_ign_gazebo_assets launch external objects implementation
--------------------------------------------------------------------------------------------
***Files to be changed***

* ``launch/launch_simulation.py``: Pass models to function ``world_bridges()``. Add ``launch_arguments`` into this function to get models to ``world_bridges.py`` file.
* ``launch/model_bridges.py``: change name to be specific for drones.
* ``src/ign_assets/model.py``: Make model class parent class, create two new model subclasses, one for drones, one for external objects. Each of the subclasses will contain specific bridges for each type of model. Also, register each definition of model for each subclass. 
* ``launch/world_bridges.py``: make it launch so the bridges return by the call to object subclass of model launch with the specific object defined in it.

RPS98, 2023-02-20:
@javilinos is it solved in #145 ?

javilinos, 2023-03-13:
@RPS98 yes, solved in #145 . Some enhancements can be done in the future though.

closed 2023-03-13 by javilinos