aerostack2/aerostack2 #179: [as2_python_api] Long wait services during initialization

pariaspe, 2023-03-08:
### Feature Description

Running the following standalone code:
```python
drone_interface = DroneInterface("drone_sim_0", verbose=True)
print(drone_interface.modules)

drone_interface.shutdown()
```
that makes this result:
```
[INFO] [1678275532.162682531] [drone_sim_0.drone_sim_0_interface]: Starting drone_sim_0
[INFO] [1678275532.689869700] [drone_sim_0.drone_sim_0_interface]: drone_sim_0 interface initialized
[WARN] [1678275533.701457376] [drone_sim_0.drone_sim_0_interface]: TakeOffBehaviour Not Available
[WARN] [1678275534.711327032] [drone_sim_0.drone_sim_0_interface]: GoToWaypointBehaviour Not Available
[WARN] [1678275535.728054260] [drone_sim_0.drone_sim_0_interface]: FollowPathBehaviour Not Available
[WARN] [1678275536.737072246] [drone_sim_0.drone_sim_0_interface]: LandBehaviour Not Available
{'takeoff': <as2_python_api.modules.takeoff_module.TakeoffModule object at 0x7efd79beb850>, 'go_to': <as2_python_api.modules.go_to_module.GoToModule object at 0x7efd79beb6d0>, 'follow_path': <as2_python_api.modules.follow_path_module.FollowPathModule object at 0x7efd79beb7c0>, 'land': <as2_python_api.modules.land_module.LandModule object at 0x7efd79be92a0>}
```

## Profiling:

[image]

Almost the **98% of the time** the code is sleeping (`time.sleep()`).

Should be consider reducing sleeping times? @miferco97 @RPS98 

### Implementation Considerations

- At drone_interface_base init method (really needed?):
https://github.com/aerostack2/aerostack2/blob/374c14feb20e6087ad879275496b196de00f427d/as2_python_api/as2_python_api/drone_interface_base.py#L107-L115

- At behavior_handler init method (change to optional? reduce default timeout?):
https://github.com/aerostack2/aerostack2/blob/374c14feb20e6087ad879275496b196de00f427d/as2_python_api/as2_python_api/behavior_actions/behavior_handler.py#L83-L90

miferco97, 2023-03-08:
The question is if we really can reduce this sleep times. Because, I understand that the wait for service will wait unit the dds_discovery finds the connection to the service server.

pariaspe, 2023-03-08:
> The question is if we really can reduce this sleep times. Because, I understand that the wait for service will wait unit the dds_discovery finds the connection to the service server.

Most of the times we could. Also, while developing it is frustrating to wait 5 secs each time a test is run... That's why I suggesting change it to optional.

miferco97, 2023-03-08:
What would happen then? it is not going to wait ?

miferco97, 2023-03-08:
> If the server (or service) is found immediately, the sleep time is interrupted.
This is not happening now? 
 
> Another option is moving the wait for availability to the start call, instead of during initialization. Sounds better?
We can try it and see if the behavior is similar. If its the same, we could move this into the start call in order to avoid all the waits() happening at the same time during initialization