aerostack2/aerostack2 #186: [as2_state_estimator] Add earth_geodetic in tf tree

pariaspe, 2023-03-10:
### Feature Description

Current tf tree only works with cartesian coordinates:
```
earth --> map --> odom --> baselink
```

Including `wgs84` node in the tree will allow to transform also to/from geodetic coordinates:
```
wgs84 --> earth --> map --> odom --> baselink
```

As soon as the geodetic origin is set, the transformation between `wgs84` and `earth` will be known and it will be static. Something similar happens with `earth`-`map` transformation (see #170).

The state estimator will absorb this complexity and the rest of the nodes would work equivalently with cartesian or geodetic coordinates.

Further implementations may also support other geodetic reference frames if needed.

### Implementation Considerations

High level layers will command the controller through a new motion_reference topic (e.g. `motion_reference/geopose`).

```cpp
void ControllerHandler::ref_geopose_callback(const geographic_msgs::msg::GeoPoseStamped::SharedPtr msg) {
  // pre-checks...

  geographic_msgs::msg::GeoPoseStamped geopose_msg = *msg;
  if (!tf_handler_.tryConvert(geopose_msg, input_pose_frame_id_)) {  // see footnote 1
    return;
  }
  ref_pose_                  = geopose_msg;
  motion_reference_adquired_ = true;

  if (!bypass_controller_) controller_ptr_->updateReference(ref_pose_);
}
```

tf handler will extend `tryConvert` method:
```cpp
bool TfHandler::tryConvert(geographic_msgs::msg::GeoPoseStamped &_geopose,
                           const std::string &_target_frame) {
  // do some stuff and call convert
  geometry_msgs::msg::PoseStamped _pose = convert(_geopose, _target_frame);
}
```

and call a new `convert` implementation will receive a geopose and return a pose:
```cpp
geometry_msgs::msg::PoseStamped TfHandler::convert(const geographic_msgs::msg::PoseStamped &_pose,
                                                   const std::string &target_frame,
                                                   const std::chrono::nanoseconds timeout) {
  // footnote 2: geopose_in --> pose_in (or similar)
  geometry_msgs::msg::PoseStamped pose_out;
  tf2::doTransform(_pose, pose_out,
                   tf_buffer_->lookupTransform(target_frame, _pose.header.frame_id,
                                               _pose.header.stamp, timeout));
  pose_out.header.frame_id = target_frame;
  pose_out.header.stamp    = _pose.header.stamp;
  return pose_out;
}
```

**¹** Notice that with the given structure, `tryConvert` will fail since input and output will have now different types.
**²** [`doTransform`](http://docs.ros.org/en/indigo/api/tf2/html/namespacetf2.html#a2bf447973d0079bb3fa862b3c947e147) use the same class types for input and output.

RPS98, 2023-03-10:
An alternative solution is to keep the use of Geometry Msgs PoseStamped, and indicate when it is geodesic or cartesian in the frame_id. In this way, the motion behaviors would work with both geodesic and Cartesian, and the motion references would not have to be extended. The tranformation will be done by the controller manager.

For that, TF Handler must be expand to interpret the frame_id wgs84 as geodesic and make the transformation to cartesian.

I think it is the most efficient way to not have to duplicate behaviors implementations, like having the Go To GPS and the Go To Cartesian. On the negative side, we would be encoding GPS information in a Pose message.

miferco97, 2023-03-10:
> An alternative solution is to keep the use of Geometry Msgs PoseStamped, and indicate when it is geodesic or cartesian in the frame_id. In this way, the motion behaviors would work with both [...]

I think that we should not abuse the message types, I mean, we should not write a geodesic coordinates into a PoseStamped message.