aerostack2/aerostack2 #188: [as2_motion_controller] Controller motion references are updated only when arrives

RPS98, 2023-03-13:
### Feature Description

The controller is currently working:
- Receives a motion reference
- Looks at its timestamp and converts it to that time.
- In parallel, the control compute output is called with a timer.

The problem: if the motion reference is in an inertial frame, such as with respect to a moving object, the reference is converted when it is received and is no longer updated until the next one arrives.

The proposal: should the controller convert the motion references constantly, not only when they are received?

RPS98, 2023-03-13:
What do you think? @pariaspe @miferco97 

Also, keep in mind #63, because when it is in hover mode, something similar happens with the hover reference.

miferco97, 2023-03-13:
I think they should update the reference too. In this case a new behavior should not be needed

RPS98, 2023-03-13:
Some alternatives:
- Perform the conversion in the main, compute output timer.
- Perform the conversion with another timer in parallel.

The change is that previously the conversion was performed at the time set in the timestamp of the motion reference. However, this change would require performing the conversion at the last available time in the TF buffer.

The problem is the overhead added by the continuous conversion in the main loop at high frequency.

miferco97, 2023-03-13:
> The change is that previously the conversion was performed at the time set in the timestamp of the motion reference. However, this change would require performing the conversion at the last available time in the TF buffer.

Hmmm yes but if you wanted for example to go a concrete point in a concrete moment of time you cannot update it continously, for example if you fix a target that was 1 meter in front of the /base_link frame, if you keep updating it then the uav will no stop

pariaspe, 2023-03-13:
Another possible scenario, if you command a go_to to a certain position but your estimation is really bad at that time, you are not updating the reference though your estimation might improve.

Shouldn't `base_link` frame be treated as a special case?

miferco97, 2023-03-15:
> Another possible scenario, if you command a go_to to a certain position but your estimation is really bad at that time, you are not updating the reference though your estimation might improve.

In this case i think that the reference should change ( be sent again)

> Shouldn't `base_link` frame be treated as a special case?

I wouldn't do this becase is not only about `base_link`, `camera_link` or `lidar_link` or any link that is not "fixed"