aerostack2/aerostack2 #196: [as2_behaviors_motion] Follow Reference Behavior

RPS98, 2023-03-21:
### ROS2 Version

humble

### Version

main

### What happened?

## Description

When launch motion_behaviors_launch, follow reference behavior displays the text on the screen: "Could not get state from frame earth to the desired frame"

## Reproduce the error

Using project ignition, launch simulation and main launcher, and see the behavior screen output. 

### Relevant log output

```shell
file:///home/rafa/Pictures/Screenshots/Screenshot%20from%202023-03-21%2011-53-44.png
```

javilinos, 2023-03-21:
Hi @RPS98, there was a redundant-missplaced warning which should be indeed caught by the exception, it is now fixed.

RPS98, 2023-03-21:
The problem is that this sentence is call every twist state callback

https://github.com/aerostack2/aerostack2/blob/298fdfa92fbc4cb3cf70b24b50e52272e23a3a49/as2_behaviors/as2_behaviors_motion/follow_reference_behavior/include/follow_reference_behavior/follow_reference_behavior.hpp#L265

closed 2023-03-22 by pariaspe