aerostack2/aerostack2 #230: [as2_state_estimator] ground truth plugin needs to support gps

javilinos, 2023-04-26:
### Feature Description

It is necessary that ground truth plugin supports gps in order to set an origin and place map in the position where the drone takes off relative to that origin.

### Implementation Considerations

- Add gps parameters when starting the node
- As the origin is set by the simulation (IE: coordinate (0,0)) the origin should not be set automatically. That said, setting origin by service or parameter.
- Build earth->map frame with the starting gps position of the drone relative to the origin. 
- Map->odom will be the identity matrix as we are dealing with ground truth position
- As we receive earth->base_link position, calculate odom->base_link with the resepctive transformation operations.

RPS98, 2023-10-04:
@javilinos Has this issue been resolved already?

javilinos, 2023-10-04:
Solved.

closed 2023-10-04 by javilinos