aerostack2/aerostack2 #254: [as2_rl] RL API

javilinos, 2023-05-12:
### Feature Description

Given the capacities of Aerostack2, it would be very suitable to implement a reinforcement learning API with different environments by default to apply RL to control problems in simulation. It is necessary to make it usable, so a way of designing a reward function for desired behaviors in a usable way would be the main innovations. 

### Implementation Considerations

- Dessign the end to end structure from the users perspective to the lowest level
- Creating a generic environment with gimnasium API, so that the main functions (eg: reward, observations, actions) can be overwritten easily by the user.
- Model the simulated environment in Ignition
- Ability to vectorize the environment (would need to decrease cpu load of as2 execution)

javilinos, 2023-05-12:
The feature request section called 'Implementation considerations' was incorrectly used, I will leave only the main requisites that the tool **should** have regardless of the architecture @miferco97

miferco97, 2023-05-12:
Okay, @javilinos, then these are "functional requirements" or "integration requirements". @perezsaura-david  what do you think ?

tavanft, 2024-07-30:
How can I use  as2 program with PyTorch for reinforcement learning? Can you give me a simple example?

alvgaona, 2026-11-16:
Stale issue. Closing as a lot has changed throughout the years.

closed 2026-11-16 by alvgaona