aerostack2/aerostack2 #263: [as2_msgs] MissionUpdate msg new types

pariaspe, 2023-05-24:
### Feature Description

## Currently

```yml
int32 drone_id
int32 mission_id

uint8 EXECUTE = 0
uint8 STOP    = 1
uint8 APPEND  = 2
uint8 INSERT  = 3

uint8 type

string mission  # json
```

| Value | Type | Description | Drone ID | Mission ID | Mission |
| :---: | --- | --- | --- | --- | --- |
| 0     | EXECUTE | Load and start mission | Target drone | New mission ID | Mission to execute |
| 1     | STOP    | Stop and reset mission | Target drone | Target mission | - |
| 2     | APPEND  | Add elements at the end | Target drone | Target mission | New elements |
| 3     | INSERT  | Add elements at the beginning | Target drone | Target mission | New elements |

## Proposal
### Changes
- New item_id 
- `stop` renamed into `abort`.
- `stop` changed. Now stop current item and start next in mission.
- `insert` changed. Now it can be chosen where to insert. 
- `append` deleted. New `insert` removes the need of `append`

| Value | Type    | Description | Drone ID | Mission ID | Mission | Item ID |
| :-----: | --------- | ------------- | ---------- | ------------ | --------- | --------- |
| 0     | EXECUTE | Load and start mission | Target drone | New mission ID | Mission to execute | - |
| 1     | LOAD    | Load mission on interpreter | Target drone | New mission ID | Mission to load | - |
| 2     | START   | Start mission        | Target drone | Target mission | - | - |
| 3     | PAUSE   | Pause mission  | Target drone | - | - | - |
| 4     | RESUME  | Resume mission | Target drone | - | - | - |
| 5    | STOP    | Stop current item and start next | Target drone | - | - | - |
| 6    | JUMP_TO | Stop current item and start X element | Target drone | Target mission | - | Element to jump to |
| 7    | REPEAT  | Repeat last X elements | Target drone | Target mission | - | Number of elements to repeat |
| 8    | INSERT  | Add elements | Target drone | Target mission | New elements | Index where insert |
| 9     | MODIFY  | Modify item in mission | Target drone | Target mission | New element | Target element |
| 10    | REMOVE | Remove item in mission | Target drone | Target mission | - | Target element |
| 11    | RESET   | Remove mission | Target drone | Target mission | - | - |
| 12   | ABORT   | Stop and reset mission | Target drone | - | - | - |

What are your thoughts about this @RPS98 @miferco97 @alcorpas10 @rdasilva01 ?

alcorpas10, 2023-05-25:
I think that with the proposed changes, the MissionUpdate message will finally cover all possible actions in the mission, so great work.
I also see that you named the action number 6 JUMP_TO, which I believe is a better than option than the previous.

rdasilva01, 2023-05-25:
It seems it now covers all possible scenarios. I don't know if the name STOP being used to go to next item and skip the current one is the most self describing one, maybe there is other word so it doesn't get mistaken with stop a mission instead of skip element. SKIP maybe?
The difference between RESET and ABORT is that in reset the current item runs until it ends?

closed 2023-06-26 by pariaspe