aerostack2/aerostack2 #270: [mocap integration] Use MOCAP4ROS2 as the standard for mocap

miferco97, 2023-05-26:
### Feature Description

MOCAP4ROS2 (https://github.com/MOCAP4ROS2-Project) is a project leaded by Francisco Martin that tries to unify the use of different mocap systems in ros2.
I have tried it and seems to work pretty well and it integrates optitrack and VICON systems. 

### Implementation Considerations

They publishes a message with all the RigidBodies with their ID and their Pose using an specific mocap_msg (https://github.com/MOCAP4ROS2-Project/mocap_msgs/blob/master/msg/RigidBodies.msg), maybe we could consider using this instead of the `ground_truth/pose` in the mocap plugin  for the state estimator ? 

@RPS98 @pariaspe what do you think ?

pariaspe, 2023-05-26:
It sounds good to me. @javilinos may we give it a try this week?

javilinos, 2023-05-26:
@pariaspe yes, I think we should if that standarizes the uses of mocap systems, even more if @miferco97 has already tried. The question is, as that changes an essential part of the projects execution with mocap, if we should do it before recording the project videos or afterwards.

miferco97, 2023-05-26:
As far as i know, this should not modify the launching of the projects, no? because the node that is launched is specified in the session.yml no ? @javilinos

javilinos, 2023-05-26:
@miferco97 true, it doesn't change the launch, but the execution should remain the same after the changes to the node. We can record the videos with what we have now as it is more urgent, and then change the node and make sure to test it out so everything remains as it should before merging the change.

closed 2024-04-18 by javilinos