aerostack2/aerostack2 #271: [ALL] Timesync policies when working with multiple robots

miferco97, 2023-05-26:
### Feature Description

Due that as2 heavily relies on the TF tree the timestamps are very important for the correct working of the system. When performing a multiagent operation, the clock of each Computer can differs, what can give some errors. 
One of the easy examples for understanding this is related with a mocap system. If you run the mocap_node in the ground station (with its own clock) and then you publish the data into the network when each of the Computers receives them, they can have errors when using the timestamp sent by the mocap node.

### Implementation Considerations

`NTP` and `chrony` seems to be the words that are more used on this field. We can have a look into: 
- https://robofoundry.medium.com/how-to-sync-time-between-robot-and-host-machine-for-ros2-ecbcff8aadc4
- https://answers.ros.org/question/379750/time-synchronization-approaches/

All this should be properly documented for ease the development of multi-agent systems.

RPS98, 2023-05-26:
We have also used SSH time synchronization to solve this problem with the onboard computer. With this method, one device acts as an SSH clock server, and clients synchronize their machine clocks with the server

miferco97, 2024-04-18:
[image]

We should take this into consideration for real flights 

https://wiki.ros.org/ROS/NetworkSetup#Timing_issues.2C_TF_complaining_about_extrapolation_into_the_future.3F