aerostack2/aerostack2 #274: [as2_motion_controller] Parameter changes are not applied properly

krepa098, 2023-05-31:
### ROS2 Version

humble

### Version

main

### What happened?

Parameter changes at runtime are not applied properly.

Setting a parameter (e.g. `ros2 param set /uav/controller_manager trajectory_control.kp.x 1.0`) will set the appropriate value in the parameter server as can be verified by a `ros2 param dump /uav/controller_manager`. However, the actual value used by the controller is not updated and thus keeps operating with the old value.

The new parameter values are passed to the callback registered by `node->add_on_set_parameters_callback()`, however, the actual value of the parameters is then later again retrieved using `node->get_parameters(_params_list)` which yields the *old* values as the changed values are only available _after_ acceptance in the callback. 

Therefore `Plugiun::updateParams` should receive the list of parameters (`std::vector<rclcpp::Parameter>`) directly from the callback instead of `std::vector<std::string>` with a subsequent call to `node->get_parameters`, which yields the old values.

pariaspe, 2023-06-01:
Hi @krepa098, thanks for the report. We'll work on it next week. Feel free to open a pull request if you come with a solution

RPS98, 2023-06-05:
Hi @krepa098 , thank you for reporting the bug. It has already been fixed in PR #275. As you indicated, the `update_param` now receives a `rclcpp::param` instead of a `std::string`, so you will need to adapt the code