aerostack2/aerostack2 #282: Installation throws a lot of warnings

drfenixion, 2023-06-15:
### Feature Description

I mean https://aerostack2.github.io/_00_getting_started/source_install.html
Many of warning about obsolete headers and deprecated contractions.

It nice to fix that for get clear install.

Ubuntu 22, ROS 2 Humble, based on docker image tiryoh/ros2-desktop-vnc:humble

Some examples:
`Finished <<< as2_behaviors_platform [26.4s]
Starting >>> as2_keyboard_teleoperation
Finished <<< as2_keyboard_teleoperation [2.01s]
Starting >>> as2_platform_dji_osdk
--- stderr: as2_behaviors_motion
In file included from /home/ubuntu/ros2/aerostack2_ws/install/as2_core/include/as2_core/utils/tf_utils.hpp:37,
                 from /home/ubuntu/ros2/aerostack2_ws/install/as2_motion_reference_handlers/include/as2_motion_reference_handlers/position_motion.hpp:40,
                 from /home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/follow_path_behavior/plugins/follow_path_plugin_position.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/ubuntu/ros2/aerostack2_ws/install/as2_core/include/as2_core/utils/frame_utils.hpp:38,
                 from /home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/follow_path_behavior/plugins/follow_path_plugin_trajectory.cpp:40:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/ubuntu/ros2/aerostack2_ws/install/as2_core/include/as2_core/utils/tf_utils.hpp:37,
                 from /home/ubuntu/ros2/aerostack2_ws/install/as2_motion_reference_handlers/include/as2_motion_reference_handlers/position_motion.hpp:40,
                 from /home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/go_to_behavior/plugins/go_to_plugin_position.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/ubuntu/ros2/aerostack2_ws/install/as2_core/include/as2_core/utils/frame_utils.hpp:38,
                 from /home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/go_to_behavior/include/go_to_behavior/go_to_base.hpp:43,
                 from /home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/go_to_behavior/plugins/go_to_plugin_trajectory.cpp:37:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
In file included from /home/ubuntu/ros2/aerostack2_ws/install/as2_core/include/as2_core/utils/frame_utils.hpp:38,
                 from /home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/land_behavior/include/land_behavior/land_base.hpp:46,
                 from /home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/land_behavior/plugins/land_plugin_platform.cpp:41:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.h:35:2: warning: #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead [-Wcpp]
   35 | #warning This header is obsolete, please include tf2_geometry_msgs/tf2_geometry_msgs.hpp instead
      |  ^~~~~~~
/home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/land_behavior/plugins/land_plugin_platform.cpp: In member function ‘virtual bool land_plugin_platform::Plugin::own_activate(as2_msgs::action::Land::Goal&)’:
/home/ubuntu/ros2/aerostack2_ws/src/aerostack2/as2_behaviors/as2_behaviors_motion/land_behavior/plugins/land_plugin_platform.cpp:62:90: warning: ‘rclcpp::Client<ServiceT>::FutureAndRequestId::operator rclcpp::Client<ServiceT>::SharedFuture() [with ServiceT = std_srvs::srv::SetBool; rclcpp::Client<ServiceT>::SharedFuture = std::shared_future<std::shared_ptr<std_srvs::srv::SetBool_Response_<std::allocator<void> > > >]’ is deprecated: FutureAndRequestId: use .future.share() instead of an implicit conversion [-Wdeprecated-declarations]
   62 |     platform_land_future_ = platform_land_cli_->async_send_request(platform_land_request_);
      |              `

miferco97, 2023-06-20:
Hi @drfenixion, the warning related with  "tf2_geometry_msgs/tf2_geometry_msgs.hpp" is there because we still wanted to have compatibility with galactic ( we did a lot of research on this distro and that is why we wanted to keep the compatibility) but we are aware that we should change this headers in the near future. 

The one related with the services is a warning that  we should check and try to fix it, since it does not seem to be the proper way of doing this construction. @RPS98, did you manage to solve this ?

Thanks for notice these !! . If you manage to improve any part of the project we would love to review any PR, and merge it into aerostack2 :)

RonaldDutu, 2023-06-30:
Sorry for the intrusion @RPS98 @miferco97. I'm doing a master thesis in which I have to implement a coverage path planning of a swarm of drones and I committed to use Aerostack2 to do it. Can I ask you why you should maintain compatibility with Galactic version considering has reached EOL. I would like to add that I will eventually develop an application so I will contribute to your project but I prefer to have a clean install.

pariaspe, 2023-07-02:
Hi @RonaldDutu , we keep maintaining Galactic after EOL because some of the drones we are using have Jetson Nano/NX (arm64) as onboard computer.  Nowadays, we can only install Galactic binaries on them since the Jetpack (and Jetson linux) is based on Ubuntu 20.04 [[1]](https://developer.nvidia.com/embedded/jetpack). Galactic is the only distro supported on Ubuntu 20.04 on arm64 architectures [[2]](https://docs.ros.org/en/rolling/Releases/Release-Galactic-Geochelone.html).
We will move on as soon as a new Jetpack based on Ubuntu 22.04 arrives. 

Anyway, we are glad you have chosen Aerostack2 for your thesis. This is a full open source and open to contribute project, we are looking forward to your contribution!

closed 2024-04-18 by miferco97