aerostack2/aerostack2 #288: [Gazebo project] as2_behavior_tree_main-1 ERROR

drfenixion, 2023-06-18:
### ROS2 Version

humble

### Version

main

### What happened?

When i run `./launch_as2.bash -b` in Gazebo project, I get error:

```
[INFO] [as2_behavior_tree_main-1]: process started with pid [3741]
[as2_behavior_tree_main-1] [ERROR] [1687087606.296221559] [drone0.bt_manager]: "TakeOffBehavior" action server not available after waiting for 1 s
[as2_behavior_tree_main-1] terminate called after throwing an instance of 'std::length_error'
[as2_behavior_tree_main-1]   what():  basic_string::_M_create
[ERROR] [as2_behavior_tree_main-1]: process has died [pid 3741, exit code -6, cmd '/home/ubuntu/ros2/aerostack2_ws/install/as2_behavior_tree/lib/as2_behavior_tree/as2_behavior_tree_main --ros-args -r __ns:=/drone0 --params-file /tmp/launch_params_21ibjgk7'].
```

I use my own docker image (with your install instructions) based on tiryoh/ros2-desktop-vnc:humble 

[image]

pariaspe, 2023-06-19:
Hi @drfenixion, I've just tested it and I'm not able to reproduce your error. I don't need to add more time to the Takeoff action neither launch it with `-t` option. 
Could please extend the information about your system?

drfenixion, 2023-06-19:
Tried `ros2 topic pub /drone0/start std_msgs/msg/String "data: ''"` 5 times with and without -t. All times it works fine.

pariaspe, 2023-06-20:
No error on gazebo window neither? Spawn problems usually come from zombie processes in the background. Check if some gazebo bridges or similar did not die properly.

closed 2024-08-09 by RPS98