aerostack2/aerostack2 #289: [ALL] improvement install dependency system

drfenixion, 2023-06-18:
### Feature Description

I see you have doubled dependency install system. Maybe one of them was inherited from old version. Looks it need to remove bash install system because it is not a best practic and save only package.xml install.

Examples:
as2_behavior_tree/install_bt_dependencies.bash
```
#!/bin/bash

sudo apt-get update
sudo apt-get install ros-$ROS_DISTRO-behaviortree-cpp-v3 -y 
sudo apt-get install ros-$ROS_DISTRO-nav2-msgs -y 
sudo apt-get install ros-$ROS_DISTRO-nav2-behavior-tree -y 
```
as2_behavior_tree/package.xml
```
<depend>behaviortree_cpp_v3</depend>
<depend>nav2_behavior_tree</depend>
<depend>nav2_msgs</depend>
```

Doubled install system confuses developers.

drfenixion, 2023-06-19:
It is not only end bash script. It is also starting install script aerostack2_ws/src/aerostack2/installers/install_package_dependencies.py

And starting script also uses in your dockerfile 
aerostack2_ws/src/aerostack2/docker/Dockerfile
`RUN /bin/bash -c 'python3 installers/install_package_dependencies.py --repos ./installers/as2_rolling.repos -p .'`

pariaspe, 2023-06-19:
`installers` folder is quite old too. We might delete it soon, what do you think @miferco97 ?
Also, docker is work in progress, latest changes are 6 months ago. It will be great if you could open a PR with your dockerfile. Thanks!

drfenixion, 2023-06-19:
> `installers` folder is quite old too. We might delete it soon, what do you think @miferco97 ? Also, docker is work in progress, latest changes are 6 months ago. It will be great if you could open a PR with your dockerfile. Thanks!

Do you ask my dockerfile?

closed 2023-06-20 by miferco97

pariaspe, 2023-06-20:
Yes, you are building a docker image, right?

drfenixion, 2023-06-20:
> Yes, you are building a docker image, right?

Yes, I do. I will make PR.

drfenixion, 2023-06-20:
Added PR with docker
https://github.com/aerostack2/aerostack2/pull/294

PR #292 (2023-06-20): Delete install_bt_dependencies.bash
Closes #289

## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Issue(s) this addresses   | #289 |
| ROS2 version tested on | - |
| Aerial platform tested on | - |

---

## Description of contribution in a few bullet points

* Removing deprecated installation file
commit 9be62add (2023-06-20): Delete install_bt_dependencies.bash

Closes #289