aerostack2/aerostack2 #296: [gazebo project] Questions

drfenixion, 2023-06-21:
### Feature Description

Due to poor documentation, I don't quite understand how I should build my project.

`aerostack2/projects/gazebo/` Is it just demonstrator or i should build project use this?

I want to use my own URDF/SDF.
I see `aerostack2/projects/gazebo/sim_config/world_swarm.json` and it is use quadrotor_base and looks it references to
`aerostack2/as2_simulation_assets/as2_ign_gazebo_assets/models/quadrotor_base/quadrotor_base.sdf.jinja`
How i should use my own URDF/SDF?

Can i write my project use only Python (i see most of as2 writen on C++)?

P.S. It is better to add real project based on as2 as example to docs.

pariaspe, 2023-07-02:
Hi @drfenixion , sorry for the late response, June has come with a lot of strong deadlines for us.

1. As we conceive it, projects are not built (but you could compile them if you wish). Projects are just a way of gathering several configuration files and launchers. So, yes, projects are kind of "demonstrators with start scripts/configs". They are not a ROS package because the not have source code initially. 
Projects are also a starting point for newcomers to use Aerostack2 in their own applications. You can pick the project that most fit your application, add to it your new source code and convert it to a ROS package if needed.

2. Adding your SDF/URDF models to Gazebo configuration path (as environment variable `IGN_GAZEBO_RESOURCE_PATH`) will enable to our launchers in `as2_ign_gazebo_assets` to find them and spawn your models in the world. I believe that adding your models path to the env var in the Gazebo launcher of the project is the best way to do it.

3. C++ is mainly use on low level due to its high performance, but you can use your preferred language with Aerostack2.

Finally, we are concerned that there is a lot work to be done in the documentation. As you might know any help in the form of pull requests are welcome, also to the documentation.

pariaspe, 2023-09-14:
Stale, closing..

closed 2023-09-14 by pariaspe