aerostack2/aerostack2 #312: [as2_project_px4_vision] Unable to open gazebo with ./launch_as2.bash -s command

RonaldDutu, 2023-07-21:
### ROS2 Version

humble

### Version

main

### What happened?

Hello everyone, when following the tutorial to launch the example of PX4 Vision when I type the following command on terminal:
`./launch_as2.bash -s -t` the Gazebo simulation window won't start but just the teleoperation window appears. I followed the installation of PX4 platform and its dependencies. I just skipped the last part [Setting up PX4 Vision Autonomy Development Kit](https://aerostack2.github.io/_03_aerial_platforms/_pixhawk/installation.html#id6) since I don't have a physical drone of that type.

I would like to add that, with the gazebo project folder everything works fine. Also, my goal with this is to be able to launch both the px4 platform and gazebo together so that afterwards I can develop my own mission scripts.
When I type the command: `./stop.bash` it returns to the previous terminal and It seems that it gets stuck in the tmuxinator command to start gazebo.

### Relevant log output

```shell
./launch_as2.bash -s
open terminal failed: not a terminal
./launch_as2.bash: line 97: 55360 Killed                  tmuxinator start -n gazebo -p utils/gazebo.yml simulation_config=${simulation_config}
[exited]
```

RPS98, 2023-07-28:
Hi @RonaldDutu,

Sorry for the delay in response. This project is up to date and working on our end.

To analyze the issue, could you please run `./launch_as2.bash -s` and navigate through tmux to the Gazebo window, and then provide the output from it?

RonaldDutu, 2023-07-28:
Hi @RPS98 I'm not understanding. How do I navigate to the Gazebo window through tmux? When I launch the command inside the project_px4_vision folder, I get another terminal.
The problem is that the Gazebo window doesn't start when I launch the command.

RPS98, 2023-07-28:
Sorry, I thought the Gazebo window would open and fail there.

I made some changes to the project to see if it's fixed. Please, update the project and try to see if it's resolved.

Another option to launch the simulator is by running: `./utils/launch_simulator.bash sim_config/world.json`, from the root directory of the project. Try running it like this and see if it launches.

RonaldDutu, 2023-07-28:
> Hi @RonaldDutu. Sorry, I thought the Gazebo window would open and fail there.
> 
> I made some changes to the project to see if it's fixed. Please, update the project and try to see if it's resolved.
> 
> Another option to launch the simulator is by running: `./utils/launch_simulator.bash sim_config/world.json`, from the root directory of the project. Try running it like this and see if it launches.

I tried to run the command after `./launch_as2.bash -s` and It started building the things related to px4 then at some point I get the following error:
> [56/56] Linking CXX shared library libAdvancedLiftDragPlugin.so
[3/4] cd /home/ronald/px4_ws/thirdpart...s/PX4-Autopilot/build/px4_sitl_default
Not running simulation (DONT_RUN is set).
You need to have gazebo simulator installed!

The problem is that It doesn't see my gazebo fortress installation. To install gazebo, I followed the procedure [here](https://gazebosim.org/docs/latest/ros_installation). I installed ROS2 Humble and then used `sudo apt-get install ros-humble-ros-gz` to install gazebo fortress. Somehow it is not recognised.

RPS98, 2023-07-31:
Hi @RonaldDutu,

I believe there has been some confusion because we need to improve the documentation on this matter. I will try to clarify the situation.

PX4 offers two simulators: [Gazebo Simulation](https://docs.px4.io/main/en/sim_gazebo_gz/) and [Gazebo Classic Simulation](https://docs.px4.io/main/en/sim_gazebo_classic/).

**The one integrated into Aerostack2 is Gazebo Classic Simulation**. It seems you don't have it installed, which is causing the error. To install it, you should follow the [instructions](https://classic.gazebosim.org/tutorials?tut=install_ubuntu). Additionally, in [PX4 Autopilot](https://github.com/PX4/PX4-Autopilot), you can find the script Tools/setup/ubuntu.sh that will help you configure it.

Regarding the new Gazebo Ignition, Aerostack2 supports its latest LTS version [Gazebo Fortress](https://gazebosim.org/docs). However, the version compatible with PX4 Autopilot is [Gazebo Garden](https://gazebosim.org/docs/garden/install_ubuntu). So far, there hasn't been a need to add compatibility with Gazebo Garden, and we will wait for the next LTS version, [Gazebo Harmonic](https://gazebosim.org/docs/latest/releases). Hopefully, PX4 Autopilot will also adapt to that version. Nevertheless, this is an open-source project, and we are open to any contributions of this kind.

I hope I have resolved your doubts.

RonaldDutu, 2023-07-31:
@RPS98 Thanks for the reply. Now, I have a clear picture of the issue. The px4 project inside Aerostack2 accepts only Gazebo Classic. I will install that and if everything works fine, I will close the issue.

closed 2023-07-31 by RonaldDutu