aerostack2/aerostack2 #335: [as2_behaviors_trajectory_generation] Launch failing when using apt binaries

pariaspe, 2023-09-25:
### ROS2 Version

humble

### Version

v1.0.0

### What happened?

Using version v1.0.1

### Relevant log output

```shell
cvar@aca4083b9cb7:~/path_planning$ ros2 launch as2_behaviors_trajectory_generation generate_polynomial_trajectory_behavior_launch.py namespace:=drone0 use_sim_time:=true
[INFO] [launch]: All log files can be found below /home/cvar/.ros/log/2023-09-25-19-20-47-515349-aca4083b9cb7-257
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [generate_polynomial_trajectory_behavior_node-1]: process started with pid [388]
[ERROR] [generate_polynomial_trajectory_behavior_node-1]: process has died [pid 388, exit code 127, cmd '/opt/ros/humble/lib/as2_behaviors_trajectory_generation/generate_polynomial_trajectory_behavior_node --ros-args --log-level info --ros-args -r __ns:=/drone0 --params-file /tmp/launch_params_0qr4zf27'].
[generate_polynomial_trajectory_behavior_node-1] /opt/ros/humble/lib/as2_behaviors_trajectory_generation/generate_polynomial_trajectory_behavior_node: error while loading shared libraries: libdynamic_trajectory_generator.so: cannot open shared object file: No such file or directory
```

pariaspe, 2024-04-18:
> @pariaspe is this solved in the last binaries ?

No, it keeps failing at 1.0.9

alvgaona, 2026-11-16:
We have to see if it's still happening in 1.1.3 or in 2.0.0 (when we release it)