aerostack2/aerostack2 #347: [as2_pkg] <Odom error>

madinabeip, 2023-10-10:
### ROS2 Version

humble

### Version

main

### What happened?

I'm trying to take off the drone 'iris_dual_camera' using the aerial platform 'as2_platform_pixhaw' in gazebo11. When I launch all the components, everything appears to work, but when I try to move the drone, the controller shows "waiting for odometry". Upon checking the odometry topics, I discovered that there weren't publishing anything. What could be causing the problem?

### Relevant log output

```shell
[as2_motion_controller_node-7] [INFO] [1696960794.797467110] [drone0.controller_manager]: Waiting for odometry 
[takeoff_behavior_node-11] [INFO] [1696960795.502452053] [drone0.TakeOffBehavior]: RUNNING
```

RPS98, 2023-10-11:
Hi @madinabeip 

Just a couple of questions.
- Have you installed Aerostack2 from source or binaries?
- For the `as2_platform_pixhawk` platform, have you followed our [installation guide](https://aerostack2.github.io/_03_aerial_platforms/_pixhawk/installation.html#aerial-platform-px4-installation-guide)?

Currently, we are using:

- PX4 Autopilot in version [v1.14.0-rc2](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0-rc2)
- [Micro XRCE][https://github.com/eProsima/Micro-XRCE-DDS] in the main branch
- [PX4 msgs](https://github.com/PX4/px4_msgs) in the main branch
- [as2_platform_pixhawk](https://github.com/aerostack2/as2_platform_pixhawk) in the main branch

When launching the simulator using PX4 Autopilot alongside Micro XRCE, it should publish odometry data on the topic `/fmu/out/vehicle_odometry`. Additionally, if you enable GPS in Aerostack2, it should be published on the topic `/fmu/out/vehicle_gps_position`

In our example `project_px4_vision`, when running `./launch_as2.bash -s`, it launches both the simulator (using the `PX4_FOLDER` environment variable pointing to PX4 Autopilot) and Micro XRCE (via the command `MicroXRCEAgent udp4 -p 8888`), along with the rest of the Aerostack2 components.

With the Python script `python3 mission.py -s`, the example mission is launched.

Given this overview of the system, to locate the error, you could follow the following steps:

- Launch the simulator independently of Aerostack2:

1.   Launch the simulator using PX4 Autopilot:
`cd /path/to/PX4-Autopilot`
`make px4_sitl gazebo-classic`
4.   Launch Micro XRCE:
`MicroXRCEAgent udp4 -p 8888`
6.   Verify that the topics are working correctly:
`ros2 topic echo /fmu/out/vehicle_odometry`

- If it works correctly, launch it from the project `project_px4_vision`:

1. Launch Aerostack2:
`cd /path/to/project_px4_vision`
`./launch_as2.bash -s`
4. Verify that the topics are working correctly:
`ros2 topic echo /fmu/out/vehicle_odometry`

Based on the results, we can determine where the error might be located.

RPS98, 2023-10-11:
The odometry process is:

1. Gazebo publish dron odometry in:
`/fmu/out/vehicle_odometry`
2. `as2_platform_pixhawk` publish the odometry in:
`/ns/sensor_measurements/odom`
3. `as2_state_estimator` set TF and publish the odometry in:
`/ns/self_localization/pose`

Can you trace which topics are been published?

Also, the `state_estimator` console log would be useful.

madinabeip, 2023-10-11:
The 3 topics are publishing , and the state_estimator console shows this messages: 
`[as2_state_estimator_node-1] [INFO] [1697045372.913041071] [drone0.state_estimator]: Construct with name [state_estimator]
[as2_state_estimator_node-1] [INFO] [1697045372.926882619] [drone0.state_estimator]: Waiting for GPS fix to set origin`

Now, the problem is that the controller and the behaviors terminal shows this msg:

In the behavior terminal
`[WARN] [1697045764.372492432] [drone0.controller_manager]: Could not get transform: "earth" passed to lookupTransform argument fixed_frame does not exist.`

In the controller terminal:
`[ERROR] [1697045764.321805956] [drone0.controller_manager]: Could not get transform: "earth" passed to lookupTransform argument source_frame does not exist.`

RPS98, 2023-10-16:
The problem is that the State Estimator is expecting GPS, and it should be receiving it through the topic:
`/fmu/out/vehicle_gps_position`
Can you check if the PX4 Autopilot is publishing that information?

Moreover, by default, if you don't set the `-g` flag, GPS shouldn't be used, and it should work correctly. Try launching:
`./launch_as2.bash -s`

If the issue persists, could you copy the entire log from the State Estimator window to see what parameters are being passed to the node?

madinabeip, 2023-10-16:
If I run `./launch_as2.bash -s` , the gps state estimator launchs with the GPS:
`ros2 launch as2_state_estimator state_estimator_launch.py namespace:=drone0 use_sim_time:=true plugin_name:=raw_odometry plugin_config_file:=sim_config/state_estimator_gps.yaml
adrian@madport:~/workspace/src/px4/project_px4_vision$ ros2 launch as2_state_estimator state_estimator_launch.py namespace:=drone0 use_sim_time:=true plugin_name:=raw_odometry plugin_config_file:=sim_config/state_estimator_gps.yaml
[INFO] [launch]: All log files can be found below /home/adrian/.ros/log/2023-10-16-19-47-13-227822-madport-71903
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [as2_state_estimator_node-1]: process started with pid [72252]
[as2_state_estimator_node-1] [INFO] [1697478433.417296121] [drone0.state_estimator]: Construct with name [state_estimator]
[as2_state_estimator_node-1] [INFO] [1697478433.440235220] [drone0.state_estimator]: Waiting for GPS fix to set origin`

I checked it, and I discovered that when launching with the -s option, the parameters gps and simulation are erroneously set to true. However, I couldn't identify any point in the code where gps is explicitly set to true. In the launch_as2.launch file, I added a echo before calling utils/aerostack2.yml, and it correctly indicates that the gps value is false. 

I'm using the main branch of project_px4_vision without any modifications.

closed 2023-10-17 by javilinos

RPS98, 2023-10-17:
@madinabeip we have update both aerostack2 and project_px4_vision. Can you check if it works correctly?

You may have to do a [rosdep install](https://docs.ros.org/en/humble/Tutorials/Intermediate/Rosdep.html#rosdep-operation).

madinabeip, 2023-10-17:
All launches correctly; the only failure is that, in this example, the drone hasn't got the sufficient power to take off. However, by adjusting the parameters on the platform and tweaking the speed settings, I managed to get it to take off, although not entirely smoothly.