aerostack2/aerostack2 #351: Arm failure

Adrimapo, 2023-11-12:
### ROS2 Version

humble

### Version

main

### What happened?

# Gazebo Version: 
11.10.2

# Problem 
When I try to take off the drone, the offboard command usually executes successfully on the first attempt. However, when trying to arm the drone, it takes several attempts until the program manages to arm it and take off.

Observing the alphanumeric viewer, I noticed a point where the drone transitions to a landed state but instantly disarms and starts displaying the message Invalid transition: LANDED -> ARM indefinitely. Occasionally, after 5-10 minutes, the robot manages to exit this state and takes off.

What could I do to expedite this process?

### Relevant log output

```shell
[as2_platform_pixhawk_node-1] [INFO] [1699825865.511480045] [drone0.platform]: OFFBOARD_ENABLED
[as2_platform_pixhawk_node-1] [INFO] [1699825873.914920790] [drone0.platform]: Transition [ARM] : New State [LANDED]
[as2_platform_pixhawk_node-1] [WARN] [1699825880.641737243] [drone0.platform]: Invalid transition: LANDED -> ARM
[as2_platform_pixhawk_node-1] [WARN] [1699825887.908103107] [drone0.platform]: Invalid transition: LANDED -> ARM
```

pariaspe, 2023-11-12:
Hi @Adrimapo , is it PX4 logging something relevant? Also, what PX4 version are you using?

Adrimapo, 2023-11-13:
The PX4 isn't logging anything relevant, I'm using:.
- Micro-XRCE: v2.4.1
- Px4 Autopilot: v1.14.0-rc2
- as2_platform_pixhawk: main branch

RPS98, 2023-11-13:
Hi @Adrimapo 

Could you please consider using [PX4 Autopilot v1.14.0](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.14.0), the latest release?

If you're working with our [project_px4_vision](https://github.com/aerostack2/project_px4_vision), launch it using:
- `./launch_as2.bash -s`

After that, initiate the mission to arm the drone with:
- `python3 mission.py -s`

If arming takes an unusually long time, kindly check the gazebo simulator frequency and PX4 FMU topics frequencies to ensure it's not a simulator-related issue.

Adrimapo, 2023-11-13:
I have run a couple of tests with version v1.14.0. I checked the simulation frequency, and it was correct. When I place the robot in the position 0,0,0,0 in my world, it takes a short amount of time. However, if I position it at '53.462', '-10.734', '0.004', '-1.6', which is its starting point, it does take much longer. Maybe I'm setting the drone position incorrectly. 

I'm setting the drone position in the next way:
{
    "world": "absolutePath.world",
    "drones": [
        {
            "model": "iris_dual_cam",
            "name:": "drone0",
            "pose": [
                53.462,
                -10.734,
                0.004,
                -1.6
            ]
        }
    ]
}

pariaspe, 2023-11-13:
> Maybe I'm setting the drone position incorrectly.

It looks good. I'm afraid it might be an issue related with PX4 SITL. Try to disable px4 parameter COM_DISARM_PRFLT (https://docs.px4.io/v1.14/en/advanced_config/parameter_reference.html#commander) to avoid disarming if taking off takes longer than expected.

Adrimapo, 2023-11-14:
It works much better with the parameter disabled, thank you @pariaspe and @RPS98 .

closed 2023-11-14 by Adrimapo