aerostack2/aerostack2 #359: [as2_behaviors] Inconsistent behavior naming between node and class

pariaspe, 2023-11-17:
- TakeOffBehavior --> TakeoffBehavior
- GoToWaypointBehavior --> GoToBehavior

Also, what about using snake_case for node clases? We are using it at every node except from behaviors.

               There is a bug on service naming due to inconsistent node name. Node names should be snake_case but right now are CamelCase, probably following behaviors class naming.

_Originally posted by @pariaspe in https://github.com/aerostack2/aerostack2/issues/358#issuecomment-1816633398_

miferco97, 2023-11-21:
Has this been solved in #57 ?

pariaspe, 2023-11-21:
No, there is a workaround in the launcher to deal with this:
https://github.com/aerostack2/aerostack2/blob/b0a7167c433b64bd3f32a0b518b27e8c21206828/as2_behaviors/as2_behaviors_motion/launch/composable_motion_behaviors.launch.py#L100-L111

I'll try to open a PR soon

closed 2023-11-27 by miferco97