aerostack2/aerostack2 #420: [project_crazyflie] Example fails with Goal Rejected

mhl787156, 2024-02-27:
### ROS2 Version

humble

### Version

main

### What happened?

Hi! 
I was testing out the crazyflie example and I found that starting the example will not work most of the time and give you a Goal Rejected Error

i.e.
```
./launch_as2.bash -s -t
```
Will lead to the log output down below where the goal handler rejects the behaviour when running `python3 mission.py` or `python3 mission.py -s`. (This is even if I don't interact with the teleop panel at all) 

[image]

and similarly with Swarm (`./launch_as2.bash -s -t -m`) 

[image]

The only thing which does work is
```
./launch_as2.bash -s
```
(No `-t`) Then the rest of the example runs fine. (python3 mission.py -s). 

However the swarm variant 
```
./launch_as2.bash -s -m
```
Still leads to Goal Rejected issues as earlier. 

(I have tried the thing with stating the drones to `./stop.bash` but it doesnt seem to make a difference)

I also tried the gazebo example from the docs and all of that runs fine too. 

I am on Ubuntu22.04, Humble and Built from source (main) 

### Relevant log output

```shell
Take Off
[WARN] [1709043407.927432000] [cf0.cf0_interface]: Goal Rejected
Take Off done
Go to with path facing [-1.0, 1.0, 1.0]
Traceback (most recent call last):
  File "/home/mickey/ros2/aerostack2_ws/project/project_crazyflie/mission.py", line 75, in <module>
    drone_run(uav)
  File "/home/mickey/ros2/aerostack2_ws/project/project_crazyflie/mission.py", line 44, in drone_run
    drone_interface.go_to.go_to_point_path_facing(goal, speed=speed)
  File "/home/mickey/ros2/aerostack2_ws/build/as2_python_api/as2_python_api/modules/go_to_module.py", line 143, in go_to_point_path_facing
    self.__go_to(point[0], point[1], point[2],
  File "/home/mickey/ros2/aerostack2_ws/build/as2_python_api/as2_python_api/modules/go_to_module.py", line 81, in __go_to
    self.start(msg, speed, yaw_mode, yaw_angle, frame_id, wait)
  File "/home/mickey/ros2/aerostack2_ws/build/as2_python_api/as2_python_api/behavior_actions/go_to_behavior.py", line 81, in start
    return super().start(goal_msg, wait_result)
  File "/home/mickey/ros2/aerostack2_ws/build/as2_python_api/as2_python_api/behavior_actions/behavior_handler.py", line 157, in start
    raise self.GoalRejected('Goal Rejected')
as2_python_api.behavior_actions.behavior_handler.BehaviorHandler.GoalRejected: Goal Rejected
```

pariaspe, 2024-02-27:
Hi @mhl787156, 
My guess is that the project is not closing properly after first `stop`. I made some changes at the project repository. Could you pull the changes and try again? Also, make sure that there is not any gazebo or aerostack2 process alive in the background before launching

mhl787156, 2024-02-27:
Thanks for the speedy reply! 

Yes it seems to be working now! What did you change? 

You might also want to do the same changes for Crazyflie Gates as the same errors were popping up there

pariaspe, 2024-02-27:
It was this line on the gazebo tmuxinator config file:

```
on_project_exit: pkill -9 -f "gazebo"
```
https://github.com/aerostack2/project_crazyflie/commit/2ad93d41b161a90081b3afa1baf9515544dcc1f4#diff-a5b149997656008ce1199e8710ed168a93a32c46e2d02f06a34f75590ec48c64L14

I don't understand why but tmux was killing sometimes the aerostack2 interfaces to gazebo. Anyway, we are closing everything on the stop script so it is redundant. 

Glad to help! Closing this issue and also changing cf_gates project.

closed 2024-02-27 by pariaspe