aerostack2/aerostack2 #421: [as2_gazebo_assets] Using different models

mhl787156, 2024-02-28:
Hi! 

I have been playing with the project_crazyflie example and I was trying to set the "model_type" to crazyflie, but the platform errors. 

Looking through the code it seems that in `as2_gazebo_assets/src/as2_gazebo_assets/models/drone.py` you hard code a DroneTypeEnum to only have the quadrotor_base and hexrotor_base, which are currently the only allowed values to model_type. 

However I also see that `as2_gazebo_assets/model` you have a number of sdfs including the crazyflie sdf. 

So how do I spawn the crazyflie model for the simulation? 

I couldn't find any guidance on this in the docs, only that the only two models in simulation are the quadrotor_base and hexrotor_base. 

Cheers!

mhl787156, 2024-02-28:
To add more info, I added `CRAZYFLIE: crazyflie` to the DroneTypeEnum, while setting model_type to crazyflie and lo and behold the crazyflie did spawn! 

But it wasn't going to take off and the controller was complaining... so I'm assuming that there are topic mismatches if using the crazyflie model?

pariaspe, 2024-02-29:
Hi @mhl787156 ,

We are not including the crazyflie in the `DroneTypeEnum` because there are some issue in the model that cause an unstable flight. The [documentation](https://aerostack2.github.io/_03_aerial_platforms/_gazebo_simulation/index.html#config-simulation) lists the models available for gazebo.
There is no plan in the near future to fix this, you can take a look at the models in this [repo](https://github.com/bitcraze/crazyflie-simulation) and check if they made some changes that fix it.

mhl787156, 2024-03-01:
Hi @pariaspe 

Okay so I managed to connect up the crazyflie, I have the props spinning on the arm message. 
Just needed to update the robotNamespace and I think the gazebo topic names are no longer /gazebo/command/... and are now just /command/... (basically just copied over quadcopter_base) 

Things fail when I request a takeoff though, props spin but vehicle doesnt move. I'm guessing that the takeoff command isn't getting through to the model which is causing Goal Rejected errors for subsequent behaviours? 

[image]

I haven't looked into your behaviours and commands yet, would be able to point me to some places which I might want to look at? Where is the link between the behaviour and sending a controller command? 

Thanks!

mhl787156, 2024-03-01:
Just to add, I tried to call the cf0/TakeoffBehaviour manually on the cli (after arming) and I get the following as expected:
```
> ros2 action send_goal /cf0/TakeoffBehavior as2_msgs/action/Takeoff "{takeoff_height: 2.0, takeoff_speed: 0.5}"
Waiting for an action server to become available...
Sending goal:
     takeoff_height: 2.0
takeoff_speed: 0.5

Goal accepted with ID: 972eac11b0c34533a715f25ca9cf4395

Result:
    takeoff_success: false

Goal finished with status: ABORTED
```

Trying to figure out if its Aerostack side or gazebo side.
From Kimberley's issue: https://github.com/bitcraze/crazyflie-simulation/issues/57#issue-2150687102 
I'd like to try and access gazebo directly and run something like the following: 

`gz topic -t /crazyflie/gazebo/command/motor_speed --msgtype ignition.msgs.Actuators -p 'velocity:[250,250,250,250]'`

However running `gz topic` in the tmux shell, or any shell doesn't seem to find the running Gazebo instance?

pariaspe, 2024-03-04:
Hey @mhl787156 , I'm out of the office until next Wednesday. I'll take a look and test your PR when I come back, but it seems an issue while creating the tf_tree. Thanks for the debugging!

mhl787156, 2024-03-04:
No worries take your time! 

Yes you are right, it is an issue with the TF tree! Above is from when I tried changing the as2_behaviours_motion to `crazyflie/body` as base link. But I'm assuming something in the system has hardcoded a look up to `base_link` as the base link. 

In the PR i've made the simpler change of just setting the crazyflie base link to `base_link` and it removes most of the above issues!

mhl787156, 2024-03-10:
Hi @pariaspe just trying to look into the velocity issue now! 
Do you have any suggestions on how to tune the attitude or angular rates? Those most be broken to produce such a bad velocity response... 
And do you know how to streamline the testing process? Its rather tedious having to compile and bring gazebo up and down everytime! How did you export to plotjuggler?

pariaspe, 2024-03-11:
Not sure about that. The model seems to be not reactive enough but it's hard to guess. As far as I know the only way to do it is to bring up and down the simulation, since the issue is in the plugins of the crazyflie model. Try to automatize your tests using a python script. To visualize the results, I'm recording a rosbag an opening it in plotjuggler. Sorry for not being more helpful..

mhl787156, 2024-03-11:
Thats alright! Maybe i'll try to automate it. 

One thing that is confusing me though is that the takeoff is always perfect - never any oscillations until it finishes the takeoff and I think the velocity controller takes over. 

Do you know if there's something different between the takeoff and after?
e.g. if the mission.py script just takesoff and then exits. 

I see that the bitcraze repo has its own simpler PID controller plugin, if the vel controller is not working I might try swap that in somehow. Might need some guidance on where the source of that could go.

pariaspe, 2024-03-11:
I was using the following mission, it might help.
```python
#!/bin/python3

"""
mission_vel_motion_ref.py
"""

import time
import rclpy
from as2_python_api.drone_interface import DroneInterface
from as2_python_api.modules.motion_reference_handler_module import MotionReferenceHandlerModule

class DroneMotionRef(DroneInterface):
    """Drone interface node with PointGimbalModule."""

    def __init__(self, name, verbose=False, use_sim_time=False):
        super().__init__(name, verbose, use_sim_time)

        self.motion_ref_handler = MotionReferenceHandlerModule(drone=self)

    def run_test(self):
        """ Run the mission """
        self.offboard()
        self.arm()
        self.takeoff(1.0, wait=True)
        time.sleep(1.0)
        while True:
            self.motion_ref_handler.speed.send_speed_command_with_yaw_angle(
                [0.5, 0.0, 0.0], pose=None, twist_frame_id='drone0/base_link', yaw_angle=0.0)
            time.sleep(0.1)

if __name__ == '__main__':
    rclpy.init()

    uav = DroneMotionRef("drone0", verbose=False, use_sim_time=True)
    uav.run_test()
    uav.shutdown()
    rclpy.shutdown()
```

To avoid issues with outer controller (as2_motion_controller), you should set behavior to load speed plugins. Then, the mission sends speed commands and the outer controller is bypassed (input and output control mode are the same).
I don't know what is your current simulation setup, but check if the takeoff is using speed or position plugin. You can check it reading what the behavior is logging, it should print what plugin is loading.

I was recording these topics:
```
            /tf
            /tf_static
            /clock
            /<%= drone %>/platform/info
            /<%= drone %>/self_localization/pose
            /<%= drone %>/self_localization/twist
            /<%= drone %>/actuator_command/twist
            /<%= drone %>/motion_reference/pose
            /gz/<%= drone %>/cmd_vel
            /<%= drone %>/motion_reference/twist
            /<%= drone %>/ground_truth/twist
            /<%= drone %>/ground_truth/pose
```
`ground_truth` are the input for the state estimator, while `self_localization` is the output. They should be same identical, since there is no need to estimate anything, you already have the **ground truth**. There is no odometry in simulation because it is not needed. You can find attitude and angular rates in both topics.
`motion_reference` are the input for the controller and the `actuator_commands` the output. `gz/md_vel` is what the platform interface (Gz<->as2) sends to Gazebo and `actuator_commad` is what receives, should be identical too (I don't remember exactly if any transformation is performed). `platform/info` is just  recorded to check that no control mode is changed during the activation and deactivation of behaviors.

I didn't check bitcraze PID controller. Is it a gazebo plugin?

mhl787156, 2024-03-11:
Hi @pariaspe thanks for all that, I will give it a bit more of a go and get back to you in the next two weeks, on leave for a bit! 

The bitcraze controller is a gazebo plugin here, no idea how that one performs though!  https://github.com/bitcraze/crazyflie-simulation/tree/main/simulator_files/gazebo/controllers

mhl787156, 2024-03-20:
Hi @pariaspe just back to trying to get to the bottom of this. I've been trying to follow the data passing throughout your system and I've got the following, does this track? 

1. The project pollibots mission.py calls a motion_ref_handler.speed.send_speed_command_with_yaw_angles. What this does is send a twist message to the motion_reference/twist topic (as2_motion_reference_handlers/basic_motion_references.py)
2. This gets read in by the as2_motion_controller which performs a PID control depending on the mode of the control. In project crazyflie there is a sim config motion controller plugin which contains the gains for this PID controller. After the PID this gets sent out on a actuator_command/twist or pose or thrust or trajectory topic.
3. The actuator command then gets read in by the aerial platform module. In our case that aerial platform is the gazebo bridge module which translates it into a gazebo command and sends it to gazbeo on /gz/{drone name}/cmd_vel.
4. Then the gazebo multicopter velocity plugin then reads in that target cmd vel message and tries to move the drone model using the gazebo P tunings in the crazyflie sdf file. 

In which case the question I have is in step 2 and whether the PID controller is activated for our example script above sending `self.motion_ref_handler.speed.send_speed_command_with_yaw_angle` commands, and whether this might be negatively affecting anything? 

We have attempted changing the gains on the Gazebo Multicopter plugin but have had no success just yet. 

I'm thinking that if this doesnt work, as a last ditch effort I could create a new Crazyflie_Gazebo Aerial platform which uses the crazyflie_sim simple PID controller to directly control the motors on the gazebo model, bypassing the gazebo velocity control plugin.

pariaspe, 2024-03-21:
> In which case the question I have is in step 2 and whether the PID controller is activated for our example script above sending `self.motion_ref_handler.speed.send_speed_command_with_yaw_angle` commands, and whether this might be negatively affecting anything?

Step 2 is not correct. Check your `motion_controller.yaml` config, parameter `use_bypass` should be enabled. Bypass avoids to perform the PID control if in and out control modes are identical. Also, check yaw_control mode, if it is not the same it might be controlling.

> We have attempted changing the gains on the Gazebo Multicopter plugin but have had no success just yet.
>
> I'm thinking that if this doesnt work, as a last ditch effort I could create a new [...]

After saying that, I strongly believe that the issue is in the model, either on the mesh or the controller plugin. You could try to build a dummy aerial_platform, but I'm afraid you won't succeed fixing the problem.

@RPS98 any thoughts about this?

mhl787156, 2024-03-22:
Sigh maybe its time to move on then, I've started conversations that we just test with your quadrotor_base model and then either run it on the tello or crazyflie in practice. 

Thanks for the help though! Probably worth poking Kimberley at bitcraze to have a look at it again! Its crazy that a good sim for the crazyflies is hard to come by!

knmcguire, 2024-04-25:
Hi! I came accross this I see (now you have my github name as well). 

To provide more context from the side of Bitcraze: we kind of more focused on Webot simulation actually, so I'm not sure if this is possible to integrate that into aerostack. Also the simulation maintenance was mostly experimental so we haven't had much time to maintain anything other than the [getting started guide for webots](https://www.bitcraze.io/documentation/tutorials/getting-started-with-simulation/) and that is without the inclusion of ROS. We hope that we'll have more time to look into it but the problem is mostly that we don't have a single resource only on simulation and other priorities got into the way. That said, usually when someone from the community helps out with PRs to improve it, we are always happy to review and test it. 

There have been some community contributions of projects to provide full SITL together with gazebo simulations (both old and classic) which seems to work pretty good, like [sim_cf2](https://www.bitcraze.io/2024/01/simulation-sim_cf2-for-multi-agent-algorithm-development/) and [crazysim](https://www.bitcraze.io/2024/04/crazysim-a-software-in-the-loop-simulator-for-the-crazyflie/). The last one managed to use the crazyflie-simulation repo of the new gazebo to get it to fly, so there seems to be something possible there.

We can also consider something else, and just use the lee-controller integration of Gazebo to get the Crazyflie to fly without any SITL in play: https://gazebosim.org/api/gazebo/6.7/classignition_1_1gazebo_1_1systems_1_1MulticopterVelocityControl.html. This requires the commands to be send in velocity though so not sure how compatible that is with aerostack2

mhl787156, 2024-04-25:
Hi @knmcguire as far as I could see aerostack primarily uses the [gazebo lee-controller integration](https://github.com/aerostack2/aerostack2/blob/07ccab06dcce7450beb4beede8e1a31ecc6c916c/as2_simulation_assets/as2_gazebo_assets/models/crazyflie/crazyflie.sdf.jinja#L310) as its controller. As can be seen from my previous tests there's something that seems wrong with the crazyflie model itself or the lee-controller parameters and I couldn't get down to the bottom of it.  

I remembered doing a side by side comparison of the crazyflie-simulation repo and the crazyflie model seems to be very similar though... but if theres a working model in crazysim with gazebo and the lee-controller then afaik it should work here with aerostack as well.

knmcguire, 2024-04-25:
ah oke. The lee controller was not implemented in the crazyflie-simulation repo, so then it makes sense why we haven't seen this.

Crazysim uses SITL build of the firmware so it just controls the motors directly. It is not using the lee controllers, but perhaps we can check how their motormodel looks then compared to ours. If they manage to fly it with SITL then I don't see a reason why the leecontroller can't either but let's see.

I'll try to see if I have some time tomorrow to look at this issue again https://github.com/bitcraze/crazyflie-simulation/issues/6. Maybe it's good to start from scratch since the gazebo standard example for the lee controller does work normally.

pariaspe, 2024-04-26:
Hi @knmcguire, thanks for your message! As @mhl787156 we are trying to make it fly via gazebo lee-controller plugin. I'll take a look at your motor model :+1: 
@RPS98 is also trying to control it via motor commands using [as2_platform_multirotor_simulator](https://github.com/aerostack2/aerostack2/tree/main/as2_aerial_platforms/as2_platform_multirotor_simulator). We'll update here any progress.

knmcguire, 2024-04-26:
Great! It is good to have several eyes on it. Currently I'm on gazebo Fortress, and even the regular [quadcopter.sdf example](https://github.com/gazebosim/gz-sim/blob/ign-gazebo6/examples/worlds/quadcopter.sdf) doesn't seem to respond... so that doesn't help

knmcguire, 2024-04-26:
arg... I'll go cry in a corner now. Wasted an whole day on reinstalling gazebo harmonic and fortress, trying different OS between native and WSL2, only to find out that using `gz sim` without the `-r` to indicate the robot world file makes it not respond to `gz topic`. I'll make sure that this is fixed in gazebo's documentation but no luck with me today :) another day I'll try again.

knmcguire, 2024-04-26:
managed to get the lee controller to work in crazyflie-simulation repo on this PR: https://github.com/bitcraze/crazyflie-simulation/pull/59. Still need to smooth things out

knmcguire, 2024-04-27:
Sorry spoke a bit too soon! takeoff goes pretty well but as soon as you give it a twist of zero it falls down again. Also the attitude doesn't seem to go very well

mhl787156, 2024-04-27:
Haha yes! I think that was as far as I got, it takes off but then just flutters about or lands depending on the tuning parameters.... 

Tried a whole bunch and it didn't seem like they affected it much...

knmcguire, 2024-04-27:
Yeah it is not a very easy thing to solve... currently only the webots model works but the variables of the model doesn't seem to really reflect correct system ID (see this issue you started: https://github.com/bitcraze/crazyflie-simulation/issues/60)

The multicopter example of gz-sim does seem to work fine. No drop in height and almost looks like a position controller https://github.com/gazebosim/gz-sim/blob/gz-sim8/examples/worlds/multicopter_velocity_control.sdf

There really needs to be a motor control tune plugin for gazebo because this is terrible to do just by restarting the sim the entire time

pariaspe, 2024-07-17:
Fixed in #590