aerostack2/aerostack2 #436: [ignition-gazebo6] <compile error>

carmendrpr, 2024-03-13:
### ROS2 Version

humble

### Version

main

### What happened?

[image]

### Relevant log output

```shell
I have fixed it by previously downloading gazebo.
```

pariaspe, 2024-03-13:
Hi @carmendrpr , it seems that one dependency is missing (probably [ignition-fortress](https://index.ros.org/d/ignition-fortress/)). Would you open a pull request fixing it?

RPS98, 2024-04-08:
Hi @javilinos, 
Have you fixed that? We should fix it and update installation guide

javilinos, 2024-04-08:
> Hi @javilinos, 
> Have you fixed that? We should fix it and update installation guide

We should first try to reproduce the error. We tried ros-humble-base jammy image but we couldn't do it, it compiled. I suspect maybe gazebo was preinstalled for some reason (it shouldn't as it is ros-base?). Anyway it is added as dependency in source install within the docs in the meantime.

RPS98, 2024-04-08:
> We tried ros-humble-base jammy image

We could try Ubuntu 22.04 image and install ROS 2 Humble Base manually, what do you think @javilinos?

javilinos, 2024-04-08:
> > We tried ros-humble-base jammy image
> 
> We could try Ubuntu 22.04 image and install ROS 2 Humble Base manually, what do you think @javilinos?

Will do that 👍