aerostack2/aerostack2 #451: [as2_keyboard_teleoperation] PySimpleGUI needs registration

RPS98, 2024-04-16:
### ROS2 Version

humble

### Version

main

### What happened?

When following the [keyboard teleoperation guide](https://aerostack2.github.io/_06_user_interfaces/_keyboard_teleoperation/index.html), PySimpleGUI asks for registration. Is there any specific free version?

[image]

miferco97, 2024-04-17:
@RPS98 we need to force to use the pysimplegui==4.60 version

javilinos, 2024-04-17:
force it with ``` pip install pysimplegui==4.60 ``` I will update documentation aswel

miferco97, 2024-04-17:
Okay, link this issue in the documentation PR @javilinos

RPS98, 2024-04-23:
Already in the documentation

closed 2024-04-23 by RPS98