aerostack2/aerostack2 #483: Understanding question : How does aerostack close the loop in real time (like Co-simulation) ? Time synchronisation between the simulator (Gazebo Ignition) and the Aerostack code (Python interface etc)nd the

vbalaji21, 2024-04-26:
I am trying to use aerostack for a realtime simulation. I want to know how do you use close the loop in the simulation ? How do you do time synchronised simulation between the aerostack python node and the simulator ? Eg:- Do you use a flag to stop and start the simulator to synchronsie etc. Please let me know, I want to know about the framework more.

miferco97, 2024-04-29:
@javilinos is the one working on how to pause and resume the Gazebo simulator through ROS 2

javilinos, 2024-05-05:
Hi @vbalaji21, pausing and resuming simulation via ROS 2 service is pretty straight forward. You can check [branch 254: as2_gazebo_assets/bridges/bridges.py](https://github.com/aerostack2/aerostack2/blob/254-as2_rl-reinforcement-learning-tools/as2_simulation_assets/as2_gazebo_assets/src/as2_gazebo_assets/bridges/bridges.py) at the end of the file to see how its added to Aerostack2. Aditionally, check [ros_gz_bridge readme](https://github.com/gazebosim/ros_gz/tree/ros2/ros_gz_bridge) to learn how to execute ros_gz_node in order to map gz world control service to ROS 2 service via terminal.

pariaspe, 2024-06-14:
Stale, closing