aerostack2/aerostack2 #543: [as2_keyboard_teleoperation] When using speed mode, rotation speed can't we 0

RPS98, 2024-06-22:
### ROS2 Version

humble

### Version

main

### Installation

binaries

### What happened?

When using speed mode, if you increase z-axis angular speed and then you decrease this, it jumps from positive to negative.

### How to reproduce your issue?

1. Press 'a': angular velocity z = 0.1 rad/s
2. Press 'd': angular velocity z = -0.1 rad/s; and it should be 0 rad/s

javilinos, 2024-06-24:
Every parameter of speed mode works like that, its not incremental, it only sets up a defined value independently of the state of the UAV. For example, when moving forward 1m/s with &uarr; , if you then hit &darr; it will go -1m/s, not 0. Only way of 0ing is hover (space).

RPS98, 2024-06-24:
Can't fix this? Is confusing, if moving 1m/s forward and you decrease 1m/s, user will expect a result of 0m/s

closed 2024-07-08 by javilinos