aerostack2/aerostack2 #549: [as2_alphanumeric_viewer] Battery not displayed correctly

miferco97, 2024-06-25:
### ROS2 Version

humble

### Version

main

### Installation

source

### What happened?

Battery shall be from 100% to 0% however it always shows 1%

### How to reproduce your issue?

Launch alphanumeric_viewer in a real platform

pariaspe, 2024-06-27:
Setting fixIssue225 to false will publish values between 0-1 (instead of 100), but also changes how battery discharges. More info [here](https://github.com/gazebosim/gz-sim/issues/225).
https://github.com/gazebosim/gz-sim/blob/055e969bba59bc8b28510a0b67263e7f9ccf8cce/src/systems/battery_plugin/LinearBatteryPlugin.cc#L622-L625

miferco97, 2024-06-28:
@javilinos please create an issue for handling this in gazebo

closed 2024-06-28 by miferco97

javilinos, 2024-06-28:
> @javilinos please create an issue for handling this in gazebo

Done #562