aerostack2/aerostack2 #571: [as2_pkg] Can not launch Gazebo Single drone example, reference frame errors<title>

Cristian-wp, 2024-07-02:
### ROS2 Version

humble

### Version

main

### Installation

source

### What happened?

I have followed the setup instruction, and launch the simulation with the bash script launch_as2.bash, but when Gazebo try to start I see the default world "blinking" and no drone model.
[image]

If I try to run a mission in the mission tmuxwindow:
````
ctrazzi@aerostack:~/aerostack2_ws/src/project_gazebo$ python3 mission.py
````
I get the following errors in the tmux controlller window:
````
[as2_motion_controller_node-1] [ERROR] [1719937948.435466307] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'drone0/base_link' and 'earth' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937948.487078619] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'drone0/base_link' and 'earth' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [WARN] [1719937948.549080000] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937948.600308948] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'drone0/base_link' and 'earth' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937948.600405914] [drone0.controller_manager]: Failed to convert command twist to output frame, from drone0/odom to drone0/base_link
[as2_motion_controller_node-1] [WARN] [1719937948.662534057] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [WARN] [1719937948.713874063] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937948.776265333] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'drone0/base_link' and 'earth' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937948.848615831] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'drone0/base_link' and 'earth' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [WARN] [1719937948.900511369] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [WARN] [1719937948.962287862] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937949.024689926] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'drone0/base_link' and 'earth' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [WARN] [1719937949.076593080] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [INFO] [1719937949.076686914] [drone0.controller_manager]: Bypassing controller
[as2_motion_controller_node-1] [INFO] [1719937949.078165033] [drone0.controller_manager]: input_mode:[SPEED YAW_SPEED BODY_FLU_FRAME ]
[as2_motion_controller_node-1] [INFO] [1719937949.078207205] [drone0.controller_manager]: output_mode:[SPEED YAW_SPEED BODY_FLU_FRAME ]
[as2_motion_controller_node-1] [INFO] [1719937949.078307475] [drone0.controller_manager]: input_pose_frame_id:[drone0/base_link]
[as2_motion_controller_node-1] [INFO] [1719937949.078333723] [drone0.controller_manager]: input_twist_frame_id:[drone0/base_link]
[as2_motion_controller_node-1] [INFO] [1719937949.078345256] [drone0.controller_manager]: output_pose_frame_id:[drone0/base_link]
[as2_motion_controller_node-1] [INFO] [1719937949.078354171] [drone0.controller_manager]: output_twist_frame_id:[drone0/base_link]
[as2_motion_controller_node-1] [INFO] [1719937949.078494106] [drone0.controller_manager]: Waiting for motion reference
[as2_motion_controller_node-1] [ERROR] [1719937949.234158634] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'drone0/base_link' and 'earth' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937949.234254432] [drone0.controller_manager]: Failed to convert reference twist to input frame, from earth to drone0/base_link
[as2_motion_controller_node-1] [ERROR] [1719937949.336828403] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937949.388416194] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937949.451142266] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'drone0/base_link' and 'earth' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937949.524435204] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937949.586417589] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937949.730421844] [drone0.controller_manager]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[as2_motion_controller_node-1] [ERROR] [1719937949.730529049] [drone0.controller_manager]: Failed to convert command twist to output frame, from drone0/base_link to drone0/base_link

````
As you can see there is a problem with the reference frame, but I think the main reason is something regarding Gazebo since the simulation enviroment is not correct... 

### How to reproduce your issue?

I use Ubuntu 22.04 on VMWare Virtual Machine, Gazebo Fortress, ROS2 Humble.
I use version 1.0.9 of aerostack2 downloaded with git clone --recursive https://github.com/aerostack2/aerostack2.git and installed from source.
I have install ROS2 Humble Complete, then I run these commands:
````
sudo apt install git -y #version (1:2.34.1-1ubuntu1.11)
sudo apt install python3-rosdep -y #version (22.0.2+dfsg-1ubuntu0.4)
sudo apt install python3-pip -y #version (0.3.0-1)
sudo apt install python3-colcon-common-extensions -y #version (0.24.0-1)

mkdir -p ~/aerostack2_ws/src/ && cd ~/aerostack2_ws/src/
git clone --recursive https://github.com/aerostack2/aerostack2.git #main 
````
Now I have install [Gazebo fortress](https://gazebosim.org/docs/fortress/install_ubuntu) from this page .

````
cd ~/aerostack2_ws
sudo rosdep init
rosdep update
rosdep install -y -r -q --from-paths src --ignore-src

sudo apt install tmux -y #version (3:20211215-1)
sudo apt install tmuxinator -y #version (3.0.1-1)
sudo apt install iputils-ping -y #version (3.2a-4ubuntu0.2)

pip install PySimpleGUI-4-foss

cd ~/aerostack2_ws/src
git clone --recursive https://github.com/aerostack2/project_gazebo
cd ~/aerostack2_w
colcon build --symlink-install
````
I did not have errors after compile the workspace.
Now I have source it and lauch the bash script.
````
cd ~/aerostack2_ws/
source install/setup.bash
cd ~/aerostack2_ws/src/project_gazebo
./launch_as2.bash -t
````

pariaspe, 2024-07-03:
Hi, it seems to be an issue with your Gazebo installation. Reference frames are built from gazebo to ros using bridges, so you might be having that warnings due to a Gazebo malfunction.
Try to run Gazebo standalone and check that is working well.

Cristian-wp, 2024-07-03:
Hi @pariaspe the problem was related to my VM. I had to disable the Accelerated 3D graphics, now is working perfectly with the same workspace I had use to open this issue.
Thank you for your dupport.

closed 2024-07-03 by Cristian-wp

AbhinavPeri, 2024-07-16:
### ROS2 Version

humble
### Version

main
### Installation

source
### What happened?

I'm getting the same errors shown above, which prevent me from running the example scripts. The installation seemed to go pretty well at first. I'm able to see the models in the simulator, and the majority of the navigation stack starts up fine. The main issue though is that I'm not able to run any of the behaviors or example scripts for gazebo. I get the same tf2 errors in the console that were shown above, and I can't launch any behaviors. I've tried building from source using gazebo harmonic and fortress, but end up with the same problem. 

Upon examining the tf2 tree, I can in fact see that the trees are separated. One tree starts with drone0 and contains all the physical links and joints from the sdf model. The other tree is rooted at earth containing the map, odom, and base_footprint frames. The base_link frame seems to be the issue, because i can see that it is being published as a child of both drone0 and odom. It seems like drone0 should actually be the base_link frame, which would join both trees very nicely. 

Upon digging further, it seems like this is due to the way the transforms are published. The codebase seems to use the PosePublisher plugin for publishing the link transforms in the sdf. This corresponds to the transforms in the tree rooted at drone0. There are some other files that publish transforms from earth to map to odom and etc. These seem to be from the state estimation nodes. The conflict between these two systems seem to be causing this bug.

How would I go about fixing this? Is this a bug in the codebase, or have I installed something incorrectly. I'm inclined to believe the latter, because I've been trying to run the tests through colcon unsuccessfully. 

```
Starting >>> as2_cli
Starting >>> as2_gazebo_classic_assets
Finished <<< as2_cli [0.02s]
Finished <<< as2_gazebo_classic_assets [0.02s]
Finished <<< as2_msgs [0.82s]           
Starting >>> as2_core
Finished <<< as2_core [5.18s]           
Starting >>> as2_behavior
Starting >>> as2_motion_reference_handlers
Starting >>> as2_gazebo_assets
Starting >>> as2_alphanumeric_viewer                                                                                      
Starting >>> as2_behavior_tree
Starting >>> as2_external_object_to_tf
Starting >>> as2_platform_multirotor_simulator
Starting >>> as2_realsense_interface
Starting >>> as2_state_estimator
Starting >>> as2_usb_camera_interface
Starting >>> as2_geozones
Starting >>> as2_map_server
Finished <<< as2_behavior_tree [0.07s]
Finished <<< as2_behavior [2.44s]                                                                                                                               
Starting >>> as2_behaviors_perception
Starting >>> as2_behaviors_platform
Starting >>> as2_behaviors_path_planning
Finished <<< as2_alphanumeric_viewer [2.58s]                                                                                                                                 
Finished <<< as2_external_object_to_tf [2.83s]                                                                                                                                         
Finished <<< as2_map_server [2.96s]                                                                                                                                                  
Finished <<< as2_usb_camera_interface [2.98s]                                                                                                                                        
Finished <<< as2_geozones [2.98s]
Finished <<< as2_realsense_interface [3.08s]
Finished <<< as2_motion_reference_handlers [3.47s]                                                                                                                               
Starting >>> as2_python_api
Starting >>> as2_behaviors_motion
Starting >>> as2_behaviors_trajectory_generation
Starting >>> as2_motion_controller
Finished <<< as2_behaviors_platform [2.48s]                                                                                                                                   
Finished <<< as2_behaviors_perception [2.62s]                                                                                                                                
Finished <<< as2_behaviors_path_planning [2.66s]
Finished <<< as2_state_estimator [5.09s]                                                                                                                           
Finished <<< as2_behaviors_trajectory_generation [3.03s]                                                                                                            
Finished <<< as2_gazebo_assets [6.93s]                                                                                                                              
Starting >>> as2_platform_gazebo
Finished <<< as2_motion_controller [3.67s]                                                                                                                              
Finished <<< as2_behaviors_motion [4.51s]                                                                                                                         
Finished <<< as2_platform_gazebo [2.01s]                                                                                            
Finished <<< as2_platform_multirotor_simulator [9.22s]                                                 
[... cut ...]
test/test_flake8.py::test_flake8
  Warning: SelectableGroups dict interface is deprecated. Use select.

-- Docs: https://docs.pytest.org/en/stable/warnings.html
---
Finished <<< as2_python_api [33.3s]	[ with test failures ]
Starting >>> as2_keyboard_teleoperation
Finished <<< as2_keyboard_teleoperation [1.17s]             
Starting >>> aerostack2
Finished <<< aerostack2 [0.05s]

Summary: 26 packages finished [44.3s]
  1 package had stderr output: as2_python_api
  1 package had test failures: as2_python_api

```

pariaspe, 2024-07-17:
Hi @AbhinavPeri ,
I would say that you might have a misconfiguration on the launched nodes (or at least the state_estimator one). I'm checking the current configuration of the project. I would let you know if I find something.

pariaspe, 2024-07-18:
Hi @AbhinavPeri , just found the issue. It seems that we change one parameter name at the state_estimator (`use_ignition_tf` --> `use_gazebo_tf`) that caused the unconnected tf tree. I'm updating the project, feel free to test it

AbhinavPeri, 2024-07-19:
Sounds good. I'll pull and check it out today. Where was this parameter set?

AbhinavPeri, 2024-07-19:
Just found the parameter you mentioned. Updating it looks like it fixes the issue. Thanks so much for the help!