aerostack2/aerostack2 #580: How can I use the Web GUI?

Cristian-wp, 2024-07-08:
Hello,
I am trying to use your package with the tutorial. For what I see on your pubblication and presentation, there is a Web GUI that can be used for path planning.

[image]

How can I use it in my simulations? I can not find any reference in the repositories.

RPS98, 2024-07-12:
Hi @Cristian-wp,

We are currently working on its release for the next version of Aerostack2, so it will be available soon

Cristian-wp, 2024-07-12:
@RPS98 can you give me an approximative release date(next month, semptember, 2025,ecc...)?

miferco97, 2024-07-12:
During this month

Cristian-wp, 2024-07-12:
@miferco97 Thank you for the quick reply 👍🏻  . It will be the same version with PX4+Gazebo(fortress) examples?

RPS98, 2024-09-25:
Hi @Cristian-wp,

Checkout this documentation: https://github.com/aerostack2/as2_web_gui/issues/5

closed 2024-09-25 by RPS98