aerostack2/aerostack2 #587: Launch Gazebo simulation separately

Cristian-wp, 2024-07-10:
Hello, I am trying to launch separately the Gazebo-Classic simulation with PX4.
Now I have manage to launch any other "TMUX winwow" separately

- Window0: Gazebo simulation (NOT WORKING)
`gazebo -p /home/ctrazzi/px4_ws/src/project_px4_vision/utils/gazebo.yml simulation_config=/home/ctrazzi/px4_ws/src/project_px4_vision/sim_config/world.json & wait`

- Window1: `MicroXRCEAgent udp4 -p 8888`
- Window2: `ros2 launch as2_platform_pixhawk pixhawk_launch.py namespace:=drone0 use_sim_time:=true platform_config_file:=/home/ctrazzi/px4_ws/src/project_px4_vision/sim_config/platform_config.yaml`
 
[image]

- Window3: `ros2 launch as2_state_estimator state_estimator_launch.py namespace:=drone0 use_sim_time:=true plugin_name:=raw_odometry plugin_config_file:=/home/ctrazzi/px4_ws/src/project_px4_vision/sim_config/state_estimator_odom.yaml`
[image]

- Window4: `ros2 launch as2_motion_controller controller_launch.py namespace:=drone0 use_sim_time:=true plugin_name:=pid_speed_controller 
motion_controller_config_file:=/home/ctrazzi/px4_ws/src/project_px4_vision/sim_config/motion_controller.yaml plugin_config_file:=/home/ctrazzi/px4_ws/src/project_px4_vision/sim_config/pid_speed_controller.yaml`
[image]

- Window5: `ros2 launch as2_behaviors_motion motion_behaviors_launch.py namespace:=drone0 use_sim_time:=true follow_path_plugin_name:=follow_path_plugin_position go_to_plugin_name:=go_to_plugin_position takeoff_plugin_name:=takeoff_plugin_position land_plugin_name:=land_plugin_speed`
[image]

- Window6: `python3 mission.py -s`

The problem with simulation consists in the fact that Gazebo does not start at all... It did not show any output error at all, simply I launch the command and the process termitates clean after some seconds...
Can you tell me how to perform correctly this action?

pariaspe, 2024-07-10:
It seems that you are not using the right gazebo launch command. Try something like: 
```bash
${AEROSTACK2_PATH}/as2_simulation_assets/gazebo_assets/scripts/default_run.sh <config-file>
```

Anyway, Gazebo11 is no longer maintained and we will soon remove `as2_gazebo_classic_assets`. Is there any reason why you are not using newer versions of Gazebo?

Cristian-wp, 2024-07-10:
@pariaspe I am using Gazebo11 only because your PX4 example where build upon it.
Is possible to useyour[ PX4 simulation example](https://aerostack2.github.io/_02_examples/pixhawk/project_px4_vision/index.html) with the new version?
There is a setting to do or there is specific branch?

pariaspe, 2024-07-10:
The example needs to be updated, but there is a [project](https://github.com/aerostack2/project_demo/tree/main/platforms_config/px4_sitl) working with the new version.

Cristian-wp, 2024-07-10:
Hi @pariaspe I have already write in the other issue, but the link is broken or is a priveate reposirory.

[image]

closed 2024-11-05 by pariaspe