aerostack2/aerostack2 #588: Swarm with Aerostack2 and PX4

Cristian-wp, 2024-07-10:
Hello, is possible combine your examples for swarm and PX4 using Gazebo-Classic?
There is any done example to do this? (Maybe with BT)

pariaspe, 2024-07-10:
There were a way to do it, but I'm not sure if it is still working. As said before, we will soon remove the package from aerostack2 and switch to Gazebo Fortress/Harmonic.

You might try this simulation description file:
```json
{
    "world": "empty",
    "drones": [
        {
            "model": "iris_fpv",
            "name": "drone0",
            "pose": [ 0.0, 0.0, 0.0, 0.0]
        },
        {
            "model": "iris_fpv",
            "name": "drone1",
            "pose": [ 2.0, 0.0, 0.0, 0.0]
        }
    ]
}
```

Cristian-wp, 2024-07-10:
Ok, but it will be still possible to use PX4 right?

_As said before, we will soon remove the package from aerostack2 and switch to Gazebo Fortress/Harmonic._ 
Have you and idea about the time? Some months, one year, some years... just to better understand if I can start using Gazebo Fortress/Harmonic or for the moment aerostack is configured to work with PX4 only with Gazebo Classic?

pariaspe, 2024-07-10:
Yes, aerostack2 it is right now working with PX4 and Gazebo Fortress and Gazebo Harmonic. Documentation is still pending, but can take a look at this [project](https://github.com/aerostack2/project_demo/tree/main/platforms_config/px4_sitl). It is running PX4 SITL inside a docker container though

Cristian-wp, 2024-07-10:
@pariaspe there is a problem with your [link](https://github.com/aerostack2/project_demo/tree/main/platforms_config/px4_sitl), if I click on it I get the github error page...is not public?

[image]

pariaspe, 2024-07-10:
My bad, project is still private because we are still working on it. We are basically following [PX4 Gazebo simulation](https://docs.px4.io/main/en/sim_gazebo_gz/) using px4 v1.14.3 tagged version. After simulation is launched, we are running:
```bash
        - MicroXRCEAgent udp4 -p 8888
        - ros2 launch as2_platform_pixhawk pixhawk_launch.py
            namespace:=<%= drone_namespace %>
            platform_config_file:=<%= config_folder %>/config_file.yaml
            fmu_prefix:=/<%= drone_namespace %>
```

Cristian-wp, 2024-07-10:
Ok, in the afternoon I will let you know if I manage to make it work.

If is possible can you provide me the same thing with PX4 but that can work with thw [swarm](https://aerostack2.github.io/_02_examples/gazebo/project_gazebo/index.html#swarm-drones) and [behavior-tree](https://aerostack2.github.io/_02_examples/gazebo/project_gazebo/index.html#project-gazebo-simulated-behavior-tree) examples?

My final goal is to use aerostack for a swarm simulation with PX4 and Behavior-Tree.

pariaspe, 2024-07-10:
My suggestion is to get it working with gazebo first (no px4) and then switch to PX4, in one or two weeks we might have PX4 swarm simulation ready. About BT, there is no plan to work on it in the near roadmap.

Cristian-wp, 2024-07-10:
I had already manage to make the examples linked above works. But I need to inteface them with PX4. 
To be more clear...I need to simulate a swarm that fly using coverage algorithm, but all the drones will be based on PX4 fly controller. The BT will be used to coordinate the action of the drones (arm, takeoff, start mission,...,disarm).

So your suggestion is to wait the end of this month and try with your new examples?

pariaspe, 2024-07-10:
Try to get the BT working first, then switching to PX4 should be easy.

closed 2024-11-05 by pariaspe