aerostack2/aerostack2 #589: Using BehaviorTree.CPP

Cristian-wp, 2024-07-10:
Hello, if I undestand correctly this project is based on BehaviorTree.CPP 3.
There is any roadmap or "trick" to make it work with BehaviorTree.CPP 4?
Is possible to use [Groot](https://www.behaviortree.dev/groot)? If yes there is any example of this?

pariaspe, 2024-07-10:
Hi, changing to BehaviorTree.CPP 4 is in the long-term roadmap, we don't enough resources now to work on the migration. If you are interested, we encourage you to work on it, we will kindly review and test your PR.
And yes, with version 3 we were using Groot.

Cristian-wp, 2024-07-10:
Hi @pariaspe you are using Groot2 right?
I have try to run BT your example with groot visualizer but I can manage to make it work...
I know this is a really dummy question, but can you please tell/show me how to load your example BT with Groot?

pariaspe, 2024-07-10:
No, we were using first version of Groot. The example got a bit outdated, I'm sorry to say that it might not work.
What's the issue? Are both loading tree and leaves?

Cristian-wp, 2024-07-10:
I have Groot2 at the moment but the editor is compatible with v3 and v4 [**"Compatible with both BT.CPP 3 and 4"**](https://www.behaviortree.dev/groot/).
I will try again in these hours, but I had try to open it some days ago in order to follow the example mission execution and it does not even open.

If I can not manage to use Groot2 I will try Groot1 in the afternoon.
I will let you know soon as possible.

AbhinavPeri, 2024-07-23:
I'm also getting stuck on this. What is the overall process for developing behavior trees and running them with aerostack2. How do I use groot to develop and run behaviors like the one in the example?

Additionally, how do I trace BT Nodes to their respective actions/services in the code? I'm having a lot of trouble figuring out how things work in the source code. I see two launch files being run in the BT example: the behavior tree itself and the behaviors. The behavior tree seems to be the one taking in the xml file and ticking various BT nodes. The behaviors launch file seems to be loading in plugins for all the allowed behaviors. How do these things communicate though, because I don't understand how the BT Nodes defined in the behavior_tree package are linked to the behaviors being started up.

Some additional questions:
- I see that there is a follow path/trajectory generation behavior. How do you use those in the behavior tree?
- Since behaviors are simply action servers or service servers, can I just call them and expect them to function normally?
- Slight bug, but I'm noticing that if I try to run the behavior tree example more than once, the offboard behavior will fail and the tree never starts executing. This is for the gazebo harmonic, Ubuntu 22.04, ros2 humble, source installation. It seems like there is some kind of timing issue that's preventing the behavior tree from running properly

pariaspe, 2024-07-24:
Hi @AbhinavPeri , I will try to explain aerostack2 working with BT as best as possible. But it's been a while since last time a worked with BT.

> How do I use groot to develop and run behaviors like the one in the example?

I used to use Groot1. You need to load (or manually create) the pool of available Tree Nodes Model (mainly Actions but also Conditions, Decorators, etc). There is one [file](https://github.com/aerostack2/aerostack2/blob/main/as2_behavior_tree/resource/actions.xml) I used in the past and has some basic actions to interact with Aerostack2.

> Additionally, how do I trace BT Nodes to their respective actions/services in the code?

Read carefully this [entry](https://www.behaviortree.dev/docs/ros2_integration) at BehaviorTree.CPP documentation. You need to understand that every leaf from the tree is wrapping a Action Client, which sends a goal to a Action Server when the leaf is ticked. Take a look also at [ROS2 Actions](https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Actions/Understanding-ROS2-Actions.html).
What might be confusing is naming around the word `Behavior`. On one hand, Aerostack2 behaviors are an extension of ROS2 actions with some extra features (there are some insights about them at our [documentation page](https://aerostack2.github.io/_04_robot_behaviors/index.html)). On the other hand, BehaviorTree.CPP behaviors are those NodeTrees or leaves I mentioned earlier.
So answering your question, Behavior (Action) Clients wrapped under NodeTree leaves are placed in code at [as2_behavior_tree](https://github.com/aerostack2/aerostack2/tree/main/as2_behavior_tree) package while Behavior (Action) Servers are placed in several packages depending on their purpose (take a look at [as2_behaviors folder](https://github.com/aerostack2/aerostack2/tree/main/as2_behaviors).

> How do these things communicate though, because I don't understand how the BT Nodes defined in the behavior_tree package are linked to the behaviors being started up.

They communicate through ROS2 actions, I am referring once again to the same [page](https://docs.ros.org/en/jazzy/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Actions/Understanding-ROS2-Actions.html).

> I see that there is a follow path/trajectory generation behavior. How do you use those in the behavior tree?

You need to use Follow Path BT Node, which uses a [port specialization](https://github.com/aerostack2/aerostack2/blob/9ad6409245339f146e7c6f1a87a07a1521f112a3/as2_behavior_tree/include/as2_behavior_tree/port_specialization.hpp#L83) to translate from string (what you insert at the leaf itself) to a vector of PoseWithID ros2 msg. It seems to be hardcoded to only two waypoints, but feel free to modify it. Syntax might be a bit weird too: `x0;y0;z0|x1;y1;z1`. 

> Since behaviors are simply action servers or service servers, can I just call them and expect them to function normally?

Yes, behaviors are just ros2 action servers but a bit more powerful. The main difference is that you pause, resume and modify the running goal, while standard action severs can only be activated and cancelled. And yes, we use Aerostack2 behaviors all the time without using BehaviorTrees. The BT is the "task planner" while the behavior system is the "executor". You can replace BT with other planners as our [as2_python_api](https://github.com/aerostack2/aerostack2/tree/main/as2_python_api) ([doc](https://aerostack2.github.io/_09_development/_api_documentation/temp_ws/src/as2_python_api/docs/source/modules.html)).

> Slight bug, but I'm noticing that if I try to run the behavior tree example more than once, the offboard behavior will fail and the tree never starts executing. This is for the gazebo harmonic, [...]

Do you mind to open a new issue describing this? I will take a look at it later.

Sorry for the long response I hope it clears up your doubts. We should create an entry at the documentation page explaining how to deal with BT using aerostack2.

pariaspe, 2024-07-26:
>  Looking at Aerostack2, it looks like the BT Nodes exposed in the actions.xml are of type BTServiceNode and BTActionNode. These are the wrappers around the Action clients you were talking about correct?

Yes, those are the wrappers I was talking about.

> Why is this different from the RosActionNodes mentioned in the BT docs you linked?

Because we built them when RosActionNodes didn't exist back in time with BehaviorTree.CPP v3. We will update them when we address migration to new version 4.

> What capabilities for path planning, obstacle avoidance, mapping, and etc are available in this repository? I've seen various different packages, but I'm not sure how to use them.

We are working on that direction right now. Most of the packages are in a work in progress status to be honest. We built the architecture for path planning and mapping and we are now filling them with new plugins and capabilities. 

> With the FollowPath behavior, it seems like you are just taking in waypoints. Is there a way to pass in velocity and acceleration feedforwards as well?

No, there is not right now. But, as I said before we are working on that direction.

> I know there is a big release coming up with better integration into PX4 and various other updates. Do you have an approximate timeline on when that release will come out?

It should be released last week. There are still pending a couple of issues, but it will be released really soon (hopefully next week).

PS: Feel free to close the issue once all your doubts are solved.

closed 2024-08-07 by pariaspe