aerostack2/aerostack2 #594: aerostack2 - as2_behavior_tree error during execution

Cristian-wp, 2024-07-16:
### ROS2 Version

humble

### Version

main

### Installation

source

### What happened?

Hello, today I have install a fresh ubuntu 22 on my PC (no virtual machine) so I have create a fresh workspace following the tutorial.
Now I am trying to run the [Gazebo BHT Example](https://aerostack2.github.io/_02_examples/gazebo/project_gazebo/index.html#project-gazebo-simulated-behavior-tree) but I get always this output error when I start the simulation with `python3 mission_behavior_tree.py` insdite TMUX:

```
[INFO] [as2_behavior_tree_main-1]: process started with pid [41997]
[as2_behavior_tree_main-1] [1721140420.253]: SequenceStar              IDLE -> RUNNING
[as2_behavior_tree_main-1] [1721140420.253]: WaitForEvent              IDLE -> RUNNING
[as2_behavior_tree_main-1] [1721140432.774]: AlwaysSuccess             IDLE -> SUCCESS
[as2_behavior_tree_main-1] [1721140432.774]: WaitForEvent              RUNNING -> SUCCESS
[as2_behavior_tree_main-1] [1721140432.774]: AlwaysSuccess             SUCCESS -> IDLE
[as2_behavior_tree_main-1] [1721140432.784]: Arm                       IDLE -> RUNNING
[as2_behavior_tree_main-1] [1721140432.787]: Arm                       RUNNING -> SUCCESS
[as2_behavior_tree_main-1] [1721140432.797]: Offboard                  IDLE -> RUNNING
[as2_behavior_tree_main-1] [1721140432.803]: Offboard                  RUNNING -> SUCCESS
[as2_behavior_tree_main-1] [1721140432.803]: TakeOff                   IDLE -> RUNNING
[as2_behavior_tree_main-1] [INFO] [1721140435.893667335] [drone0.bt_manager]: result
[as2_behavior_tree_main-1] [1721140436.394]: TakeOff                   RUNNING -> SUCCESS
[as2_behavior_tree_main-1] [1721140436.394]: GoTo                      IDLE -> RUNNING
[as2_behavior_tree_main-1] You (maybe indirectly) called BT::convertFromString() for type [unsigned char], but I can't find the template specialization.
[as2_behavior_tree_main-1] 
[as2_behavior_tree_main-1] [INFO] [1721140447.565287352] [drone0.bt_manager]: result
[as2_behavior_tree_main-1] You (maybe indirectly) called BT::convertFromString() for type [unsigned char], but I can't find the template specialization.
[as2_behavior_tree_main-1] 
[as2_behavior_tree_main-1] [1721140447.565]: GoTo                      RUNNING -> SUCCESS
[as2_behavior_tree_main-1] [1721140447.565]: GoTo                      IDLE -> RUNNING
[as2_behavior_tree_main-1] [1721140447.662]: GoTo                      RUNNING -> FAILURE
[as2_behavior_tree_main-1] [1721140447.662]: GoTo                      FAILURE -> IDLE
[as2_behavior_tree_main-1] [1721140447.662]: SequenceStar              RUNNING -> FAILURE
[as2_behavior_tree_main-1] [1721140447.662]: SequenceStar              FAILURE -> IDLE
[INFO] [as2_behavior_tree_main-1]: process has finished cleanly [pid 41997]

```

And the drone in Gazebo after the error remain still at midair.

[image]

What can be the problem?

### How to reproduce your issue?

I have follow [Gazebo BHT Example](https://aerostack2.github.io/_02_examples/gazebo/project_gazebo/index.html#project-gazebo-simulated-behavior-tree).

At the moment I have this packages version in my workspaces and with `colcon build --symlink-install` I get no error in output.

- git clone --recursive https://github.com/aerostack2/aerostack2.git **# commit c50d7a9b11b3b61e7f65e6823de2d4136cdc6c62**
- git clone --recursive https://github.com/PX4/px4_msgs.git -b release/1.14 **# commit ffb6e80e1c17e5714395611a020c282a87af8fa4**
- git clone --recursive https://github.com/aerostack2/as2_platform_pixhawk.git **# commit 7fd04fb101457140757304e6d465b3425f013eb6**
- git clone --recursive https://github.com/aerostack2/project_gazebo.git **#commit 137eaa0a1267e02fc83d6cf5fc626480cdb0d45f** 

This is my .bashrc:
```
############################################
#	ROS HUMBLE
############################################
source /opt/ros/humble/setup.bash

############################################
#	aerostack2
############################################
source ~/aerostack2_ws/install/setup.bash
```

pariaspe, 2024-07-19:
Hi @Cristian-wp, I've just tested this on main and it is working. Warnings about `BT::convertFromString()` appear but the drone finishes the mission correctly. Should we close this issue?

[image]

Cristian-wp, 2024-07-19:
No, I have manage to make it work only with this commit: 4d4261bc5a76df1f89df0d4d0b55d3aa55c890fb 

With the main branch I have always the same behavior and dont end the mission.