aerostack2/aerostack2 #615: [as2_state_estimator] Transform between odom and earth with Ground Truth plugin is not as expected

RPS98, 2024-07-26:
### ROS2 Version

humble

### Version

main

### Installation

binaries

### What happened?

Expected TF tree:
- Earth at world origin
- Map at initial UAV pose
- Odom at initial UAV pose

When using `raw_odometry` and not GPS origin:
- Earth at initial UAV pose
- Map at initial UAV pose
- Odom at initial UAV pose

When using `raw_odometry` and GPS origin:
- Earth at GPS origin
- Map at initial UAV pose using GPS transform
- Odom at initial UAV pose using GPS transform

When using `ground_truth` and `mocap_pose`:
- Earth at world/mocap origin
- Map at initial UAV pose
- Odom at initial UAV pose

Current TF tree when using `ground_truth` plugin:
- Earth at world/mocap origin
- Map at world/mocap origin
- Odom at world/mocap origin

### How to reproduce your issue?

Launch project_gazebo and open rviz:

[image]

Drone0 starts at [0.5,0.0,0.0], but odom and map frames starts at the same point of earth frame

RPS98, 2024-07-26:
@javilinos @miferco97 What do you think?

miferco97, 2024-07-26:
[image]
It seems like the part of the code in charge of generating this transform has been commented :S

javilinos, 2024-07-26:
When I updated the code in this [commit](https://github.com/aerostack2/aerostack2/commit/72c482fcfadcc3af3769596d0e2e31447ba7c96f#diff-5a063a1fe52022e938589eb2615e868f355cae920995294432e31194e7d4ca4b) for the map to be generated in the starting position I managed to build the transformation tree as desired. However, when testing it in simulation, controller would fail for some reason, so I reverted back in the next [commit](https://github.com/aerostack2/aerostack2/commit/7e96753300fc140d7b13d8e0c69290e2fe0d040c) (It was quite a while ago actually). In theory, only uncommenting those lines would make it right. I would test it again and see if it works as intended.

javilinos, 2024-07-26:
As I said only by uncommenting the lines I made back in the day, it already works as expected:

[image]

About the changes to the set_origin_on_start parameter I would open another Issue where we can discuss how we want that parameter to work/be modified

closed 2024-07-29 by RPS98