aerostack2/aerostack2 #634: New Aerostack2 release date

Cristian-wp, 2024-08-07:
Hello, there are any news or maybe a date about the new version release?

pariaspe, 2024-08-13:
> Hello, there are any news or maybe a date about the new version release?

Hi, release is done and binaries will be available in next Humble sync (probably in one or two weeks)

pariaspe, 2024-09-01:
Release is out!
https://discourse.ros.org/t/preparing-for-humble-sync-2024-08-28/39316/2

closed 2024-09-01 by pariaspe

Cristian-wp, 2024-09-02:
@pariaspe thank you for the reply! I have already see it, but I have not found the [Web User Interface](https://github.com/aerostack2/aerostack2/issues/580#issuecomment-2225151628) reported in your paper...If is in your new release how can I launch it?

pariaspe, 2024-09-03:
No, it wasn't released this time. We had some difficulty packing the WUI and we had a big delay in the release, so we decided to continue without it. @RPS98 might guide you on how to source install and launch it.

Cristian-wp, 2024-09-10:
Hi @pariaspe , thank you for your reply. I hope it will be fixed soon.
@RPS98 can you please guide me to source install the version with GUI?