aerostack2/aerostack2 #638: [as2_simulation_assets] GZ Harmonic launch issues with AS2 1.1.0 built from source

botmayank, 2024-08-11:
### ROS2 Version

humble

### Version

main

### Installation

source

### What happened?

There don't seem to be any errors in the terminal but Gazebo doesn't launch from the launch scripts mentioned on https://aerostack2.github.io/_03_aerial_platforms/_gazebo_simulation/index.html#installation

using:
```bash
# Source AS2
source /media/mayank/HDD1/3_MJ_Projects/5_ROS_Aerial/2_Indoor_Nav/2_Aerostack2/aerostack2_ws/install/setup.bash --extend

# go to project gazebo repo
cd /media/mayank/HDD1/3_MJ_Projects/5_ROS_Aerial/2_Indoor_Nav/2_Aerostack2/project_gazebo

# Platform launch
ros2 launch as2_platform_gazebo platform_gazebo_launch.py namespace:=drone0 simulation_config_file:=sim_config/world.json

# Simulation launch
ros2 launch as2_gazebo_assets launch_simulation.py headless:=true verbose:=true run_on_start:=false  use_sim_time:=true simulation_config_file:=sim_config/world.json
```

[image]
[image]

However, when I launch Gazebo manually and then open any project, the drone then shows up properly. 

Another gap I've observed in the docs above is that the parameter `config_file` should be `simulation_config_file` in the commands for launch. 

I do want to mention that my install seems to be fine since when I launch via the `project_gazebo` [repo's](https://github.com/aerostack2/project_gazebo) launch scripts, things are launching perfectly and teleop also works using `./launch_as2.bash -t`

### How to reproduce your issue?

I use Ubuntu 22.04, Gazebo Harmonic, ROS2 Humble and followed
[Aersostack GZ Harmonic Setup](https://aerostack2.github.io/_03_aerial_platforms/_gazebo_simulation/index.html#installation)

and [Aerostack Source Install](https://aerostack2.github.io/_00_getting_started/source_install.html#getting-started-ubuntu-installation-source)

botmayank, 2024-08-11:
@miferco97 Opened https://github.com/aerostack2/aerostack2.github.io/pull/73

RPS98, 2024-08-12:
Hi @botmayank,

It seems that when you launch Gazebo, you set the `headless` parameter to `true`. This means that the graphical interface will not be launched. If you prefer to have a graphical interface, you can set `headless` to `false`, which is the default value.

However, can you verify if the drone completes the mission correctly using RViz or the alphanumeric viewer?

botmayank, 2024-09-01:
@RPS98 thanks for pointing it out :facepalm:  :face_in_clouds: Everything looks fine. Also, thanks for merging the PR I opened for fixing the docs
https://github.com/aerostack2/aerostack2.github.io/pull/73

Closing the issue. :sweat_smile:

closed 2024-09-01 by botmayank