aerostack2/aerostack2 #641: [as2_pkg] Crazyflie simulation model characteristics are not correct

knmcguire, 2024-08-14:
### ROS2 Version

humble

### Version

main

### Installation

source

### What happened?

Hi!

I just wanted to let you know that the Crazyflie simulation model has incorrect characteristics that has been introduced by this PR: https://github.com/aerostack2/aerostack2/pull/608. The crazyflie weighs now 1.5 kilos and standard variables of the multicopter plugins have been taken that are meant for much bigger platform.

Of-course it depends of how the simulation is used for. For higher level autonomy this is probably not a problem but for lower leve control it will be because the dynamics aren't accurate. 

### How to reproduce your issue?

This is just an observation. The crazyflie simulation github repo uses the system ID parameters from the crazyflie and has been tuned to have stable flight: https://github.com/bitcraze/crazyflie-simulation

RPS98, 2024-08-16:
Hi @knmcguire,

Thank you very much for the information.

I have tried updating our model with the parameters mentioned at https://github.com/aerostack2/aerostack2/tree/641-as2_pkg-crazyflie-simulation-model-characteristics-are-not-correct

The problem we are encountering is that the MulticopterVelocityControl is not accurate.
- When the reference is v = [0.0, 0.0, 0.0] m/s, the drone moves at v = [0.0, 0.0, 0.16] m/s (approximate).
- When the reference is v = [0.0, 0.0, 0.5] m/s, the drone moves at v = [0.0, 0.0, 0.7] m/s (approximate).

Are you experiencing the same issue?

As you mentioned, we have been using the simulation to test autonomous missions, so it was more important for us that the drone moved accurately rather than having realistic dynamics. However,  we would like to achieve both.

Thank you very much!

knmcguire, 2024-10-28:
Hi! I'm very sorry that I missed your question there, but yes there seems to be a discrepency in the motion velocity control plugin.... Perhaps an external ROS plugin for PID control for ROS will just work better here.