aerostack2/aerostack2 #646: Add Sensors to drone

carlosmatac, 2024-08-22:
Hi! I'm trying to add a camera to a drone changing the world.json file. The new camera appears in the simulation and also the topic. But I believe no information is being published there. Does anyone else have this problem? Is there anything else I need to change? Maybe I need to trigger something to make the drone activate the camera? Thank you!

pariaspe, 2024-08-22:
Hi @carlosmatac , could you please give some details about your settings? Are you using binary (`apt`) or source installation? Could you copy the `world.json` you are using?

carlosmatac, 2024-08-22:
I'm using binary install. This is the world.json:
{
    "world_name": "empty",
    "drones": [
        {
            "model_type": "quadrotor_base",
            "model_name": "drone0",
            "xyz": [
                1.0,
                1.0,
                0.3
            ],
            "rpy": [
                0,
                0,
                0.0
            ],
            "flight_time": 60,
            "payload": [
                {
                    "model_type": "hd_camera",
                    "model_name": "camera0"
                }
            ]
        }
    ]
}

pariaspe, 2024-08-22:
Binary release is currently v1.0.9, where the is a bug on sensors. Rename your `model_name` with the same name of your `model_type`, this should temporary work until new version v1.1.1 is synced. 
```json
"payload": [
                {
                    "model_type": "hd_camera",
                    "model_name": "hd_camera"
                }
            ]
```

carlosmatac, 2024-08-22:
Awesome, it works, thank you!

closed 2024-08-22 by pariaspe