aerostack2/aerostack2 #655: [as2_pkg] <FollowPath client not working correctlyt in C++>

carlosmatac, 2024-09-09:
### ROS2 Version

humble

### Version

main

### Installation

binaries

### What happened?

I'm trying to write a ROS2 client in C++ for **FollowPath behavior**. I've already wirtten and tried the clients for the rest of the behavior and it worked perfectly but this is not happening with the FollowPath behavior.
```cpp
void DroneInterfaceNode::sendFollowPathGoal(const as2_msgs::action::FollowPath::Goal &aGoal) {
    if (!mpFollowPathClient->wait_for_action_server(std::chrono::seconds(5))) {
        RCLCPP_ERROR("Action server not available after waiting");
        return;
    }

    auto send_goal_options = rclcpp_action::Client<as2_msgs::action::FollowPath>::SendGoalOptions();
    send_goal_options.result_callback = [this](const rclcpp_action::ClientGoalHandle<as2_msgs::action::FollowPath>::WrappedResult & result) {
        if (result.code == rclcpp_action::ResultCode::SUCCEEDED) {
            RCLCPP_INFO("FollowPath succeeded");
        } else {
            RCLCPP_ERROR("FollowPath failed");
        }
    };

    // Setting goal feedback 
    send_goal_options.feedback_callback = [this](rclcpp_action::ClientGoalHandle<as2_msgs::action::FollowPath>::SharedPtr, 
                                                 const std::shared_ptr<const as2_msgs::action::FollowPath::Feedback> feedback) {
        RCLCPP_INFO("Waypoints to target: %u",feedback->remaining_waypoints);
        
        
    };
    mpFollowPathClient->async_send_goal(aGoal, send_goal_options);
}
```
When I set the goal and send it to the function, the drone starts to move but it only goes to the first point in the path and then stops. Nevertheless I keep receiving feedback, telling me that the remaining points are decreasing. Here's an example of an output:
```bash
[INFO] [1725890694.378146586] [DroneInterface]: Starting follow path
[INFO] [1725890694.381939041] [DroneInterface]: Published feedback
[INFO] [1725890697.110601500] [DroneInterface]: Waypoints to target: 10
[INFO] [1725890697.211043225] [DroneInterface]: Waypoints to target: 10
[INFO] [1725890697.341237550] [DroneInterface]: Waypoints to target: 10
[INFO] [1725890697.435603790] [DroneInterface]: Waypoints to target: 9
[INFO] [1725890697.559259552] [DroneInterface]: Waypoints to target: 8
[INFO] [1725890697.641176070] [DroneInterface]: Waypoints to target: 7
[INFO] [1725890697.741313553] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890697.869558033] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890697.992308782] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.074882745] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.172189809] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.274772807] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.431026057] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.531425834] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.635869143] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.736112961] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.897149441] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890698.948049130] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890699.081485053] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890699.168595135] [DroneInterface]: Waypoints to target: 6
[INFO] [1725890699.301638646] [DroneInterface]: Waypoints to target: 5
[INFO] [1725890699.376968563] [DroneInterface]: Waypoints to target: 4
[INFO] [1725890699.480490639] [DroneInterface]: Waypoints to target: 3
[INFO] [1725890699.615560291] [DroneInterface]: Waypoints to target: 2
[INFO] [1725890699.718987315] [DroneInterface]: Waypoints to target: 1
```
**Does anybody know what the problem could be?** Thank you :)

### How to reproduce your issue?

I use Ubuntu 22.04, Gazebo Fortress, using project ...

miferco97, 2024-09-09:
Hi Carlos, can you provide the console log from the Behaviors nodes, if you are using our proyects shall be in the behaviors tmux tab.

carlosmatac, 2024-09-09:
**Logs:**
```bash
[follow_path_behavior_node-1] [WARN] [1725893999.692233328] [drone0.FollowPathBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[go_to_behavior_node-2] [WARN] [1725893999.705556166] [drone0.GoToBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[takeoff_behavior_node-4] [WARN] [1725893999.741553195] [drone0.TakeoffBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[go_to_behavior_node-2] [WARN] [1725893999.798351042] [drone0.GoToBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[takeoff_behavior_node-4] [WARN] [1725893999.802446619] [drone0.TakeoffBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[go_to_behavior_node-2] [WARN] [1725893999.849805897] [drone0.GoToBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[takeoff_behavior_node-4] [WARN] [1725893999.853688272] [drone0.TakeoffBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[land_behavior_node-3] [WARN] [1725893999.854300507] [drone0.LandBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[follow_path_behavior_node-1] [WARN] [1725893999.923559315] [drone0.FollowPathBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[takeoff_behavior_node-4] [WARN] [1725893999.930234600] [drone0.TakeoffBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
```
**Launch file**
```bash
#!/bin/bash

usage() {
    echo "  options:"
    echo "      -b: launch behavior tree"
    echo "      -m: multi agent"
    echo "      -r: record rosbag"
    echo "      -t: launch keyboard teleoperation"
    echo "      -n: drone namespace, default is drone0"
}

# Arg parser
while getopts "bmrtn" opt; do
  case ${opt} in
    b )
      behavior_tree="true"
      ;;
    m )
      swarm="true"
      ;;
    r )
      record_rosbag="true"
      ;;
    t )
      launch_keyboard_teleop="true"
      ;;
    n )
      drone_namespace="${OPTARG}"
      ;;
    \? )
      echo "Invalid option: -$OPTARG" >&2
      usage
      exit 1
      ;;
    : )
      if [[ ! $OPTARG =~ ^[wrt]$ ]]; then
        echo "Option -$OPTARG requires an argument" >&2
        usage
        exit 1
      fi
      ;;
  esac
done

source utils/tools.bash

# Shift optional args
shift $((OPTIND -1))

## DEFAULTS
behavior_tree=${behavior_tree:="false"}
swarm=${swarm:="false"}
record_rosbag=${record_rosbag:="false"}
launch_keyboard_teleop=${launch_keyboard_teleop:="false"}
drone_namespace=${drone_namespace:="drone"}

if [[ ${swarm} == "true" ]]; then
  simulation_config="sim_config/world2.json"
  num_drones=2
else
  simulation_config="sim_config/world.json" 
  num_drones=1
fi

# Generate the list of drone namespaces
drone_ns=()
for ((i=0; i<${num_drones}; i++)); do
  drone_ns+=("$drone_namespace$i")
done

for ns in "${drone_ns[@]}"
do
  tmuxinator start -n ${ns} -p tmuxinator/session.yml drone_namespace=${ns} simulation_config=${simulation_config} behavior_tree=${behavior_tree} &
  wait
done

if [[ ${record_rosbag} == "true" ]]; then
  tmuxinator start -n rosbag -p tmuxinator/rosbag.yml drone_namespace=$(list_to_string "${drone_ns[@]}") &
  wait
fi

if [[ ${launch_keyboard_teleop} == "true" ]]; then
  tmuxinator start -n keyboard_teleop -p tmuxinator/keyboard_teleop.yml simulation=true drone_namespace=$(list_to_string "${drone_ns[@]}") &
  wait
fi

tmuxinator start -n gazebo -p tmuxinator/gazebo.yml simulation_config=${simulation_config} &
wait

# Attach to tmux session ${drone_ns[@]}, window mission
tmux attach-session -t ${drone_ns[0]}:mission
```

miferco97, 2024-09-09:
please provide logs from the rest of the tabs, specially the ones that shows ERRORs or WARNs

carlosmatac, 2024-09-10:
**Solved!** I was sending the path without initializing the ID of each waypoint :)
```bash
[WARN] [1725952787.861733315] [drone0.FollowPathBehavior]: Could not get transform: Could not find a connection between 'earth' and 'drone0/base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[INFO] [1725952787.930159298] [drone0.FollowPathBehavior]: START
[ERROR] [1725952787.930224601] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930241703] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930261480] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930273693] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930291767] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930312365] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930323837] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930335238] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930350998] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[ERROR] [1725952787.930362639] [drone0.FollowPathBehavior]: Behavior reject, waypoint id is empty
[INFO] [1725952787.930381365] [drone0.FollowPathBehavior]: Follow path goal accepted
[INFO] [1725952787.930394770] [drone0.FollowPathBehavior]: Follow path with speed: 1.000000
``

closed 2024-09-10 by pariaspe