aerostack2/aerostack2 #662: [as2_python_api] Sensors/Camera api in the python api

singhsegv, 2024-09-20:
### ROS2 Version

humble

### Version

main

### Installation

binaries

### What happened?

This is just a doubt instead of a bug/feature. I am new to the aerostack2 platform, trying to implement a YOLO based autonomous drone on Tello. My doubt is regarding the python api, is there a way to access camera data from the python api?

I couldn't find any and my guess is to use `sensor_measurements/camera` topic to get and process the frames. Am I in the correct direction?

### How to reproduce your issue?

N/A

pariaspe, 2024-09-23:
Hi @singhsegv ,
Yes, you can. You need to change your simulation description file:
```json
{
    "world_name": "empty",
    "drones": [
        {
            "model_type": "quadrotor_base",
            "model_name": "drone0",
            "xyz": [ 1.0, 1.0, 0.3 ],
            "rpy": [ 0, 0, 0.0 ],
            "payload": [
                {
                    "model_name": "front_camera",
                    "model_type": "hd_camera"
                }
            ]
        }
    ]
}
```

You can find some details about the [simulation description file](https://aerostack2.github.io/_03_aerial_platforms/_gazebo_simulation/index.html#config-simulation) at the documentation file.
I'm closing this issue, but feel free to keep the conversation open.

closed 2024-09-23 by pariaspe

singhsegv, 2024-09-24:
Hey @pariaspe , my issue is not related to any platform specific implementation at all. I want to understand is there a way to get camera frames from the aerostack2 framework's python API or not. One way around is to spin up a ros node and subscribe to the topic, but I want to understand the correct way using the python API.

The code snippet that you provided will just enable camera in the simulation, but that is not what my query is about. How to go about using the enabled camera frames in the python api is my doubt.