aerostack2/aerostack2 #663: [as2_platform_gazebo] Reset plaftform state machine when GZ Reset Srv is called

miferco97, 2024-09-23:
### Feature Description

When a GZ tries to reset, the platform's state machine does not change, so if the drone is flying, arming and launch will be rejected because the state machine is not in the start state.

### Implementation Considerations

The reset service can be called through GZ services, so a new bridge from ROS2 to GZ must be included for the reset service.

miferco97, 2024-09-24:
Gazebo crashes when the quadrotor model is removed. When you press the reset button the models loaded with the spawn function are removed and only the world is restarted. The models shall be included in the world file before spawning if we want the reset function to work propperly.

pariaspe, 2024-11-28:
Fixed in https://github.com/gazebosim/gz-sim/pull/2686 and workaround implemented in aerostack2 in #702, #699

closed 2024-11-28 by pariaspe