aerostack2/aerostack2 #672: [as2_pkg] <title>

Cristian-wp, 2024-10-09:
### ROS2 Version

humble

### Version

main

### Installation

source

### What happened?

Hi @pariaspe , I was trying to figure out how to correctly use GoToGps instead of GoTo as motion behavior with @RPS98  in [this issue](https://github.com/aerostack2/project_mavlink/issues/4#issuecomment-2402411008) and I think we discover a bug since as @RPS98 say: "The issue you are encountering is that the behavior tree is trying to perform the conversion by calling the service /drone0/geopath_to_path, which is no longer used in Aerostack2."

How can I overcome this problem?

### How to reproduce your issue?

I have use mine exagon_gps.xml tree instead of [square.xml](https://github.com/aerostack2/project_mavlink/blob/main/trees/square.xml):
```
<?xml version="1.0"?>
<root main_tree_to_execute="BehaviorTree">
    <!-- ////////// -->
    <BehaviorTree ID="BehaviorTree">
        <SequenceStar>
            <Decorator ID="WaitForEvent" result="" topic_name="start">
                <AlwaysSuccess/>
            </Decorator>
            <Action ID="GetOrigin" service_name="get_origin" />
            <Action ID="Arm" service_name="set_arming_state"/>
            <Action ID="Offboard" service_name="set_offboard_mode"/>
            <Action ID="TakeOff" height="5.0" speed="0.5"/>
            <Action ID="GoToGps" max_speed="0.2" latitude="47.3980619" longitude="8.546164" altitude="5" yaw_angle="0.0" yaw_mode="1"/>
            <Action ID="GoToGps" max_speed="0.2" latitude="47.3980169" longitude="8.5462787" altitude="5" yaw_angle="0.0" yaw_mode="1"/>
            <Action ID="GoToGps" max_speed="0.2" latitude="47.3979270" longitude="8.5462787" altitude="5" yaw_angle="0.0" yaw_mode="1"/>
            <Action ID="GoToGps" max_speed="0.2" latitude="47.3978820" longitude="8.5461640" altitude="5" yaw_angle="0.0" yaw_mode="1"/>
            <Action ID="GoToGps" max_speed="0.2" latitude="47.3979270" longitude="8.5460492" altitude="5" yaw_angle="0.0" yaw_mode="1"/>
            <Action ID="GoToGps" max_speed="0.2" latitude="47.3980169" longitude="8.5460492" altitude="5" yaw_angle="0.0" yaw_mode="1"/>
            <Action ID="Land" speed="0.5"/>
            <Action ID="Disarm" service_name="set_arming_state"/>
        </SequenceStar>
    </BehaviorTree>
    <!-- ////////// -->
    <TreeNodesModel>
        <Action ID="Arm">
            <input_port default="set_arming_state" name="service_name">Arming service name</input_port>
        </Action>
        <SubTree ID="ArmTakeoff">
            <input_port default="false" name="__shared_blackboard">If false (default), the Subtree has an isolated blackboard and needs port remapping</input_port>
            <input_port default="2" name="tk_height">Takeoff Height</input_port>
            <input_port default="0.5" name="tk_speed">Takeoff Speed</input_port>
        </SubTree>
        <Action ID="Disarm">
            <input_port default="set_arming_state" name="service_name">Arming service name</input_port>
        </Action>
        <Action ID="FollowPath"/>
        <Action ID="GoToGps">
            <input_port name="max_speed">Maximum cruise speed</input_port>
            <input_port name="latitude" default="0.0">Latitude</input_port>
            <input_port name="longitude" default="0.0">Longitude</input_port>
            <input_port name="altitude" default="0.0">Altitude</input_port>
            <input_port name="yaw_angle" default="0.0">Angle of yaw</input_port>
            <input_port name="yaw_mode" default="0">Yaw mode flag: {KEEP_YAW = 0; PATH_FACING = 1; FIXED_YAW = 2}</input_port>
        </Action>
        <Condition ID="IsFlying"/>
        <Action ID="Land">
            <input_port default="0.5" name="speed">Landing speed</input_port>
        </Action>
        <Action ID="Offboard">
            <input_port default="set_offboard_mode" name="service_name">Offboard service name</input_port>
        </Action>
        <Action ID="TakeOff">
            <input_port default="2" name="height">Takeoff Height</input_port>
            <input_port default="0.5" name="speed">Takeoff Speed</input_port>
        </Action>
        <Action ID="GetOrigin">
            <output_port name="latitude">Latitude</output_port>
            <output_port name="longitude">Longitude</output_port>
            <output_port name="altitude">Altitude</output_port>
            <input_port name="service_name" default="get_origin">Get origin</input_port>
        </Action>
        <Action ID="SetOrigin">
            <input_port name="latitude" default="0.0">Latitude</input_port>
            <input_port name="longitude" default="0.0">Longitude</input_port>
            <input_port name="altitude" default="0.0">Altitude</input_port>
            <input_port name="service_name" default="set_origin">Set origin</input_port>
        </Action>
        <Action ID="GpsToCartesian">
            <input_port name="latitude" default="0.0">Latitude</input_port>
            <input_port name="longitude" default="0.0">Longitude</input_port>
            <input_port name="z" default="0.0">Altitude</input_port>
            <output_port name="out_pose">Target Pose</output_port>
            <input_port name="service_name" default="geopath_to_path">Geopath to be converted</input_port>
        </Action>
        <Decorator ID="WaitForEvent">
            <output_port name="result">Result</output_port>
            <input_port name="topic_name">Event Topic Name</input_port>
        </Decorator>
    </TreeNodesModel>
    <!-- ////////// -->
</root>
```

This is my [aerostack2.yaml](https://github.com/aerostack2/project_mavlink/blob/main/tmuxinator/aerostack2.yaml) that I use in my project_mavlink:
```
<%
# Input parameters
drone_namespace           = @settings["drone_namespace"]
motion_controller_plugin  = @settings["motion_controller_plugin"]
rosbag                    = @settings["rosbag"] == "true"
%>
name: <%= drone_namespace %>
attach: false
root: ./
startup_window: mission_monitoring
windows:
  # Zenoh
  # - zenoh:
  #     layout:
  #     panes:
  #       - zenoh-bridge-ros2dds

  # Platform
  - platform:
      layout: tiled
      panes:
      # Platform
        - ros2 launch as2_platform_mavlink as2_platform_mavlink_launch.py
            namespace:=<%= drone_namespace %>
            platform_config_file:=config/config.yaml
        - ros2 launch as2_platform_mavlink mavros_launch.py
            namespace:=<%= drone_namespace %>
            mavros_config_file:=config/config.yaml

  # Basic Robotics Functions
  - basics_robotics_functions:
      layout:
      panes:
        # State Estimation
        - ros2 launch as2_state_estimator state_estimator_launch.py
            namespace:=<%= drone_namespace %>
            config_file:=config/config.yaml
        # Motion Controller
        - ros2 launch as2_motion_controller controller_launch.py
            namespace:=<%= drone_namespace %>
            config_file:=config/config.yaml
            plugin_name:=<%= motion_controller_plugin %>
            plugin_config_file:=config/<%= motion_controller_plugin %>.yaml

  # Behaviors
  - behaviors:
      layout:
      panes:
        # Motion Behaviors
        - ros2 launch as2_behaviors_motion motion_behaviors_launch.py
            namespace:=<%= drone_namespace %>
            config_file:=config/config.yaml
        # Trajectory Generation Behavior
        - ros2 launch as2_behaviors_trajectory_generation generate_polynomial_trajectory_behavior_launch.py
            namespace:=<%= drone_namespace %>
            config_file:=config/config.yaml

  # Mission execution
  - mission_execution:
      layout:
      panes:
        # Behavior tree
        - ros2 launch as2_behavior_tree behavior_trees.launch.py
            drone_id:=<%= drone_namespace %>
            tree:=./trees/exagon_gps.xml #/home/ctrazzi/aerostack2_ws/src/project_mavlink/trees/square.xml #./trees/square.xml

  # Mission monitoring
  - mission_monitoring:
      layout: tiled
      panes:
        - ros2 run as2_alphanumeric_viewer as2_alphanumeric_viewer_node
            --ros-args -r  __ns:=/<%= drone_namespace %>
        # To have Aerostack2 running and the mission executor for each UAV
        - ros2 run as2_python_api mission_executor --n <%= drone_namespace %> --add_namespace  
        - echo Run here the mission

  <%- if rosbag %>
  # Rosbag
  - rosbag:
      layout:
      panes:
        - ./rosbag/record_rosbag.bash <%= drone_namespace %>
  <%- end %>
```

I have use drone0 as namespace for the mission:
`python3 mission_behavior_tree.py -n drone0`

You can see other relevant information in the [issue](https://github.com/aerostack2/project_mavlink/issues/4#issuecomment-2402411008) I open in project_mavlink repository.

pariaspe, 2024-10-09:
Hi @Cristian-wp,

Currently there is no GoToGPS behaviour. However, we are able to use GPS points in python missions, because we are converting those GPS points to cartesians first. Then, calling the GoTo with the cartesian values.

We have to do the conversion somewhere, either before calling the behaviour, in the behaviour or using a tf transformation.

Cristian-wp, 2024-10-09:
Hi @pariaspe, thank for the reply.
So I have to pass my GPS waypoint to gps_to_cartesian Action then to go_to_action right?

pariaspe, 2024-10-09:
> Hi @pariaspe, thank for the reply.
> So I have to pass my GPS waypoint to gps_to_cartesian Action then to go_to_action right? 
> 

Well, kind of. There is no gps_to_cartesian action yet. We built a workaround for the as2_python_api only.

Cristian-wp, 2024-10-09:
@pariaspe I have modify previous comment.
Another question, geopath_to_path was used in Aerostack1?

pariaspe, 2024-10-09:
> @pariaspe I have modify previous comment.
> Another question, geopath_to_path was used in Aerostack1? Maybe can it be used in the 2?

I would suggest to modify the action client from the NodeTree to use the GPSHandler to get the conversion first and then call the behaviour.

I can share an example later.

As far as I know, geopath_to_path was used in a previous release of aerostack2. We decided to not continue that approach.

Cristian-wp, 2024-10-09:
@pariaspe thank you very much. I will try to go on using yout GPSHandler conversion example.

pariaspe, 2024-10-10:
So, an example of implementation would be:

```cpp
// at as2_behavior_tree/plugins/go_to_gps_action.cpp

GoToGpsAction::GoToGpsAction(
  const std::string & xml_tag_name,
  const BT::NodeConfiguration & conf)
: nav2_behavior_tree::BtActionNode<as2_msgs::action::GoToWaypoint>(
    xml_tag_name, as2_names::actions::behaviors::gotowaypoint, conf)
{
  node_ = config().blackboard->get<rclcpp::Node::SharedPtr>("node");

  get_origin_client =
    node_->create_client<as2_msgs::srv::GetOrigin>("get_origin");
}

void GoToGpsAction::on_tick()
{
  getInput("altitude", geopose.pose.position.altitude);
  getInput("latitude", geopose.pose.position.latitude);
  getInput("longitude", geopose.pose.position.longitude);

  while (!get_origin_client->wait_for_service(std::chrono::seconds(1))) {
    if (!rclcpp::ok()) {
      RCLCPP_ERROR(
        node_->get_logger(),
        "interrupted while waiting for the service. exiting.");
      return;
    }
    RCLCPP_INFO(
      node_->get_logger(),
      "service: %s not available, waiting again...",
      service_name_.c_str());
  }

  auto request = std::make_shared<as2_msgs::srv::GetOrigin::Request>();

  auto result = get_origin_client->async_send_request(request);
  if (rclcpp::spin_until_future_complete(
      node_, result,
      std::chrono::seconds(1)) !=
    rclcpp::FutureReturnCode::SUCCESS)
  {
    RCLCPP_WARN(
      node_->get_logger(),
      "failed to receive response from service '%s'",
      service_name_.c_str());
    return;
  }

  float latitude = static_cast<float>(this->future_result_.get()->origin.latitude); 
  float longitude = static_cast<float>(this->future_result_.get()->origin.longitude);
  float altitude = static_cast<float>(this->future_result_.get()->origin.altitude);

  as2::gps::GpsHandler gps_handler;  // requires #include "as2_core/utils/gps_utils.hpp"
  gps_handler.setOrigin(latitude, longitude, altitude);  // Set origin for GPS

  double x, y, z;
  gps_handler.LatLon2Local(gps_pose.pose.position.latitude, gps_pose.pose.position.longitude,
     gps_pose.pose.position.altitude, x, y, z);  // conversion

  goal_.target_pose.point.x = x;
  goal_.target_pose.point.y = y;
  goal_.target_pose.point.z = z;

  getInput("max_speed", goal_.max_speed);
  getInput("yaw_angle", goal_.yaw.angle);
  getInput("yaw_mode", yaw.mode);        // TODO(pariaspe): runtime warning, called
                            // BT::convertFromString() for type [unsigned char]
}
```

Cristian-wp, 2024-10-10:
Thank you @pariaspe , tomorrow and this afternoon I will try to use it and I let you know about it 👍🏻

Cristian-wp, 2024-10-14:
@pariaspe I am sorry but I had a problem and I was not able to test your script.
As soon as I can I let you know the results.

closed 2026-11-16 by alvgaona