aerostack2/aerostack2 #673: Robostack Compatibility?

mhl787156, 2024-10-09:
### Feature Description

Hi! 

We would very much like to use aerostack2 for teaching, but we are currently struggling with cross-platform issues and how most students only have windows or mac... 

We have previously tried docker, but its still not ideal as a teaching tool for the number of things that can go wrong. 

We have recently been looking into Robostack (https://robostack.github.io/) as a potential solution to this problem. It uses conda (bleh...) but it reportedly just works with gazebo, rviz etc! Its only downside is that it does not have the full set of packages usually available (https://robostack.github.io/humble.html)

I have been trying to build AS2 within robostack, but I've been struggling with dependencies. In particular geographiclib. I was wondering if you have tried to build within robostack? 

p.s. are you guys going to roscon? I heard you were at IMAV in Bristol where I worked previously - would be great to meet!

pariaspe, 2024-10-09:
Hi @mhl787156 ,

We are doing some teaching with Aerostack2 too. If we want to dive into ROS programming we use docker, check this [repository](https://github.com/aerostack2/demo_ROSConES24) used in a workshop a couple of weeks ago. If ROS is omitted we use [Unibotics](https://unibotics.org/) (or its offline version [RoboticsAcademy](https://github.com/JdeRobot/RoboticsAcademy)). It is mainly to teach robotic algorithms with no focus in the middleware.

We didn't try Robostack, but please share your progress if you keep trying to integrate them.

No, we are not going to ROSCON this year... And yes, @javilinos went to this IMAV, other team members went to local Spanish ROSCON. We will more opportunities to meet in the future for sure :)

mhl787156, 2024-10-10:
Hi @pariaspe thanks for the pointers! Your repo looks really good too, if we can procure some laptops for teaching I might just use that... 

In the meantime for robostack, I'm getting the following error on build:

```
--- stderr: as2_alphanumeric_viewer
CMake Error at /home/mickey/dev/aerostack2_ws/install/as2_core/share/as2_core/cmake/export_as2_coreExport.cmake:61 (set_target_properties):
  The link interface of target "as2_core::as2_core" contains:

    GeographicLib::GeographicLib_SHARED

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /home/mickey/dev/aerostack2_ws/install/as2_core/share/as2_core/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /home/mickey/dev/aerostack2_ws/install/as2_core/share/as2_core/cmake/as2_coreConfig.cmake:41 (include)
  CMakeLists.txt:28 (find_package)
```

Looking into things, it looks like the version of GeographicLib on conda is much newer (2.4) than the one provided on Ubuntu 22.04 (1.52). There have been a number of updates and I'm not sure if things are compatible anymore. 

I'm not sure how robostack works with rosdep, but it doesn't manage to grab GeographicLib

Having a look online I'm still unsure as to why GeographicLib_SHARED might be missing. Would you guys have any experience with this? 

I have tried to link it to the Ubuntu version which kind of works (with some modifications), but its not ideal as its not cross-platform.

pariaspe, 2024-10-14:
> Having a look online I'm still unsure as to why GeographicLib_SHARED might be missing. Would you guys have any experience with this?

We install it using rosdep which uses the apt version (check how the dependency is defined in [ROS Index](https://index.ros.org/d/geographiclib/)). As you said, the version installed using apt is v1.52 on Ubuntu 22.04 ([package](https://launchpad.net/ubuntu/jammy/+source/geographiclib) at launchpad). I'm sorry to say that we are forced to use that version in ROS Humble.
I would suggest to use the same version to avoid incompatibilities. However, it seems that `as2_core` is building but packages that depend on `as2_core` can't found `GeographicLib`. Weird...

mhl787156, 2024-10-18:
Okay, so I have made some additional discoveries:
 
1. the CMAKE PROJECT_DEPENDENCIES variable should include GeographicLib. The moment I include GeographicLib within this variable, all of the downstream geographiclib errors go away! I think because it wasn't in that variable, the ament_export_dependencies on line 119 wasn't eposing geographiclib

2.  When colcon building, we have to ignore as2_rviz_plugins there is some QT OpenGL issue which I don't want to get into... we don't use this package right now, so will ignore it! Command: colcon build --symlink-install --packages-ignore as2_rviz_plugins 

3. There is an issue with as2_behaviors_perception but  has an opencv version issue with the detect_aruco_markers_behaviour .

```
error: no match for 'operator=' (operand types are 'cv::Ptr<cv::aruco::Dictionary>' and 'cv::aruco::Dictionary')
   88 |   aruco_dict_ = cv::aruco::getPredefinedDictionary(cv::aruco::DICT_6X6_250);
```

Which according to online is a problem with >v4.7.0 if the code has been written for 4.5.1 (ubuntu default). It should be the case that the opencv imports in the HPP file for detect_aruco_markers_ehaviour can be updated to the new import paths, but im not sure what the new imports should be. - Maybe you've come across this if you porting to newer ROS version? 
 
If we ignore behavious perception it actually builds! Though there are some more issues to ge through to get as2 working properly. Will update if we get any closer.

pariaspe, 2024-10-19:
> 1. the CMAKE PROJECT_DEPENDENCIES variable in as2_core should include GeographicLib. The moment I include GeographicLib within this variable, all of the downstream geographiclib errors go away! I think because it wasn't in that variable, the ament_export_dependencies on line 119 wasn't eposing geographiclib

We will take a look at this, thanks for the report.

> Which according to online is a problem with >v4.7.0 if the code has been written for 4.5.1 (ubuntu default). It should be the case that the opencv imports in the HPP file for [...]

It sounds familiar, we did face some issues when trying Jazzy. I don't remember exactly which error we got.

Good to hear that you get it working! Kudos :+1: