aerostack2/aerostack2 #707: [as2_map] Map integration for planning

dvdmc, 2024-11-18:
### Feature Description

This issue discusses integrating a map interface that can be used across as2 for reactive navigation, high-level planning, etc.

### Implementation Considerations

 The initial proposal is a map plugin system with an abstract interface focused on the required methods for the planner:
- The map module is used inside other modules such as planners. The planner uses the abstract interface and loads the map plugin specified in the configuration.
- The abstract interface will include common methods like: `isOccupied`, `isVisible`, `isPathTraversable`... The abstraction can be very simple (`isOccupied`), or more complex (`distanceToObstacle`). This allows the integration of different maps exposing different functionalities to different planners such as simply checking for occupancy or performing gradient-based trajectory optimization.
- The map plugin will implement the underlying functionality for different maps: Octomap, Voxblox, GridMap, Elevation maps... It will also take care of receiving (subscribing to) the map data/updates or loading maps from files and will hold a reference of the data structure made available to the planner through the abstract interface. This is important for planners to have fast access to the map data.

[image]

pariaspe, 2024-11-19:
Hey @dvdmc,

Thanks for the suggestions, I agree with the idea of having an abstract interface shared for maps and planners. Map nodes could be external (as you suggested) or custom depending on the needs. Currently, we have a custom map node which is very simple and it could be a good starting point for testing the very first implementation of the abstract classes and making decisions about the common method needed.

So, what do you think on start working on the abstract interface first and then redo the implementation for our custom map server?

dvdmc, 2024-11-20:
Yes, I think the abstract interface should come first to define the requirements. To move forward we should answer the following questions:
- (Inmediate) Can you point me to any module for planner, low-level controller, etc. that should have access to a map to improve its functionality? I could use that one for testing the abstract interface.
- (Future) Can you list the current AS2 modules that you think could interact with maps at some point?

Also, here are some references about how the abstract interfaces can be implemented for planning:
- [OMPL](https://ompl.kavrakilab.org/api_overview.html): users have to implement "StateSpace" and "StateValidityChecker"
- [NAV2](https://docs.nav2.org/concepts/index.html#environmental-representation): check "Environmental Representation"

pariaspe, 2024-11-21:
> (Inmediate) Can you point me to any module for planner, low-level controller, etc. that should have access to a map to improve its functionality? I could use that one for testing the abstract interface.

Sure, you can take at the planners at [as2_behaviors_path_planning](https://github.com/aerostack2/aerostack2/tree/main/as2_behaviors/as2_behaviors_path_planning). A brief explanation about the package, there is one node `PathPlannerBehavior` that contains all the ros2 interfaces (action, services and topic) and there are the plugins, where the planning implementation is. Currently, there are only two plugins: `a_star` and `voronoi`.

> (Future) Can you list the current AS2 modules that you think could interact with maps at some point?

I would say that at least [as2_behaviors_path_planning](https://github.com/aerostack2/aerostack2/tree/main/as2_behaviors/as2_behaviors_path_planning) and [as2_behaviors_trajectory_generation](https://github.com/aerostack2/aerostack2/tree/main/as2_behaviors/as2_behaviors_trajectory_generation) will interact. But, we should ask other colleagues. 

Thanks for the references!

dvdmc, 2025-01-30:
Feedback on f4cf21d:
- The abstract interface above is currently specific per `planner`. However, I believe most of the functions will be the same for a kind of `planner`. Examples of shared abstract interfaces: getOccupancy, getSize(s), worldToMap (cellToPixel, gridToImg...). These can all be implemented into an occupancy map with specializations for the update. More complex functions such as getDistance are would be specific to distance fields. Each planner could have a minimum requirement but could work on more complex representations (children of the required map). Finally, I think there should be an object for the map abstraction even if it's only a wrapper for common `costmap` or `cv::Mat`. Otherwise, the abstract interface has to be implemented at the `GraphSearcher` or `Planner` level, which generates redundancy.
- The idea of the searcher is interesting since it will be common to transform the map into a simpler more "computer science" structure to solve the planning problem. However, I think there is some entanglement between `map` and `graph` concepts in the current implementation. I think the graph should be constructed, and possibly maintained, over the map (the searcher might only need a "simpler" version of the planner to solve the problem) which means that the map has preference and ownership of its data. The map should be updated with external (subscribed) updates from the map server, and the planner can use the searcher and/or the map to get information. Again, the map should be responsible for computing things related to its state (occupancy, distance to obstacles, frontiers).

pariaspe, 2025-01-31:
Hey, thanks for the feedback!

> most of the functions will be the same for a kind of planner

Agreed, I'm just half way of it. Roadmap is to create a map abstraction next and then move current methods from `planner`s to the map abstractions.

>  I think the graph should be constructed, and possibly maintained, over the map which means that the map has preference and ownership of its data.

Agreed too. Even in some controlled scenarios you can assume your map to be just a graph. It's what we are doing during windmill inspections for instance.