aerostack2/aerostack2 #709: [as2_gazebo_assets] Spawn drone from outside as2_gazebo_assets shares needs full path

mhl787156, 2024-11-22:
### ROS2 Version

humble

### Version

main

### Installation

binaries

### What happened?

Hey guys #698 and associated Pull request for spawning drone outside as2_gazebo works great! Tysm for implementing that! 
Only hiccup I've found is that when you give it a payload, e.g. hd_camera, it can't find the model as it tries to look in the parent folder. 

I just copied the quadrotor_base into my own project files for reference.

```
            {% elif sensor.model == 'hd_camera' and not sensor.gimbaled -%}

                {% include 'models/hd_camera/hd_camera.sdf.jinja' with context %}
``` 

Looked into it, you can pass multiple paths into the Jinja2 Loader, so I first locally changed `jinja_gen.py` to take a csv as `env_dir` 

```python
    env_dir = [Path(p) for p in args.env_dir.split(',')]
    env = jinja2.Environment(loader=jinja2.FileSystemLoader(env_dir))
    template = env.get_template(os.path.relpath(args.filename, env_dir[0]))
```

Then on the drone.py side:

```python
        env_dir = [str(filename.parent.parent)]

        resource_path = os.environ.get('GZ_SIM_RESOURCE_PATH')
        if resource_path:
            env_dir += [str(Path(p).parent) for p in resource_path.split(':')]
#
        command = [
            'python3',
            f'{jinja_script}/jinja_gen.py',
            filename,
            ",".join(env_dir),
            '--namespace',
            f'{self.model_name}',
            '--sensors',
            f'{payload}',
            '--battery',
            f'{self.flight_time}',
            '--output-file',
            f'{output_file_sdf}',
        ]
```

Gotchas incluuding needing to grab the parent, and the fact that you can't send posix path so need to convert on other side. 

Not sure if this is a good solution, but it seems to load it correctly now! 

### How to reproduce your issue?

I use Ubuntu 22.04, Gazebo Fortress

pariaspe, 2024-11-25:
Hi @mhl787156 ,

Well, the pipeline does not exactly go like that. On `drone.py` we first collect all possible paths where the jinja template can be located using `GZ_SIM_RESOURCE_PATH` env var. If file is not found, an exception is raised, and if it is found, we return the file itself. Then, we define the jinja2 environment as the parent folder of that file found, for later rendering the template.

https://github.com/aerostack2/aerostack2/blob/04bfd0aac537d2cdf06987aea88d3bf76d3e2675/as2_simulation_assets/as2_gazebo_assets/src/as2_gazebo_assets/models/drone.py#L221-L224

https://github.com/aerostack2/aerostack2/blob/04bfd0aac537d2cdf06987aea88d3bf76d3e2675/as2_simulation_assets/as2_gazebo_assets/src/as2_gazebo_assets/models/drone.py#L187-L209

Anyway, I'll take a look asap :+1:

pariaspe, 2024-11-28:
Hi @mhl787156 ,
I'm not sure what you tried to do. The following SDF uses a custom drone model with two cameras attached and it works for me:
```xml
<?xml version="1.0" ?>

<sdf version="1.8">
    <model name='quadrotor_dual_cam'>
        <!-- Platform base model-->
        <include merge="true">
            <uri>model://quadrotor_base</uri>
        </include>

        <!-- TFs -->
        <plugin
            filename="gz-sim-pose-publisher-system"
            name="gz::sim::systems::PosePublisher">
            <publish_link_pose>true</publish_link_pose>
            <publish_sensor_pose>true</publish_sensor_pose>
            <publish_collision_pose>false</publish_collision_pose>
            <publish_visual_pose>false</publish_visual_pose>
            <publish_nested_model_pose>true</publish_nested_model_pose>
            <publish_model_pose>false</publish_model_pose>
            <use_pose_vector_msg>true</use_pose_vector_msg>
            <static_publisher>true</static_publisher>
            <static_update_frequency>100</static_update_frequency>
        </plugin>
        <!-- Motors -->
        <!-- Not shown, see quadrotor_base  -->

        <!-- Velocity Controller -->
        <!-- Not shown, see quadrotor_base -->

        <!-- Odometry aka Ground Truth (no noise) -->
        <plugin
            filename="gz-sim-odometry-publisher-system"
            name="gz::sim::systems::OdometryPublisher">
            <dimensions>3</dimensions>
            <odom_publish_frequency>100</odom_publish_frequency>
        </plugin>
        
        <!-- IMU -->
        <plugin
            filename="gz-sim-imu-system"
            name="gz::sim::systems::Imu">
        </plugin>

        <!-- Payload: Frontal and Ventral Cameras -->
        <include>
            <name>frontal_cam</name>
            <uri>model://hd_camera</uri>
            <pose>0 0 0.05 0 0 0</pose>
        </include>
        <joint
            name="frontal_cam_joint" type="fixed">
            <parent>base_link</parent>
            <child>frontal_cam</child>
        </joint>

        <include>
            <name>ventral_cam</name>
            <uri>model://hd_camera</uri>
            <pose>0 0 -0.05 0 1.57 0</pose>
        </include>
        <joint
            name="ventral_cam_joint" type="fixed">
            <parent>base_link</parent>
            <child>ventral_cam</child>
        </joint>

    </model>
</sdf>
```
(I'm not including motors and control plugins for shortness)